koide3/direct_visual_lidar_calibration #6: Hello, I am using velodyne16, but after data preprocessing, the point cloud becomes disorganized

af-doom, 2023-04-13:
**Before opening an issue**
If the issue is about build errors:
- [ ] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [1] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [1 ] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [ ] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots and sample data**
If applicable, add screenshots and sample data to reproduce your problem.

**Environment:**
 - OS: [e.g. Ubuntu 20.04]
 - ROS version: [e.g., ROS1 noetic]

**Additional context**
Add any other context about the problem here.
[image]

koide3, 2023-04-14:
Please properly fill the items in the template and describe the problem in detail. It is difficult to find and fix the cause of the problem without sufficient information.

Ning030, 2023-06-27:
Have you solved this problem? If so, can you provide a solution? Thanks Thanks!

koide3, 2023-07-05:
Duplicated https://github.com/koide3/direct_visual_lidar_calibration/issues/13

closed 2023-07-05 by koide3