koide3/direct_visual_lidar_calibration #7: error while running direct_visual_lidar_calibration preprocess

gitissues79, 2023-04-13:
I am having an issue while running preprocess command 
rosrun direct_visual_lidar_calibration preprocess data/ process/ -a -d -v

data_path: /media/store/sony/lidar+cam_calib/data/
dst_path : /media/store/sony/lidar+cam_calib/process/
input_bags:
- /media/store/sony/lidar+cam_calib/data/_2023-04-13-17-49-57_0.bag
- /media/store/sony/lidar+cam_calib/data/_2023-04-13-17-50-27_1.bag
- /media/store/sony/lidar+cam_calib/data/_2023-04-13-17-50-57_2.bag
- /media/store/sony/lidar+cam_calib/data/_2023-04-13-17-51-28_3.bag
- /media/store/sony/lidar+cam_calib/data/_2023-04-13-17-51-57_4.bag
- /media/store/sony/lidar+cam_calib/data/_2023-04-13-17-52-27_5.bag
topics:
- camera_info:
- image      :cam0/image_raw
- points     :/vlp16_1/velodyne_points
intensity_channel: intensity
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: cam0/image_raw
Aborted (core dumped)

Environment: 
OS: ubuntu 20.04
ROS Version: noetic

koide3, 2023-04-14:
Please fill the template form when opening an issue to help us recognize your situation and problem.

gitissues79, 2023-04-17:
I have filled the template and updated the issue i am facing.

koide3, 2023-04-18:
Thanks for your cooperation. Could you share one of the rosbags so that I can reproduce the problem?

gitissues79, 2023-04-18:
Can I send it through mail?

gitissues79, 2023-04-18:
I am sorry, I cant share here.

gitissues79, 2023-06-09:
the above isuues are resolved and I am able to calibrate it, but I am getting the skewed images and distorted 
the input data/rosbag doesnt have any distortion and skewed images.
[image]

koide3, 2023-07-05:
It might be a problem of the viewer (related to OpenGL driver) rather than the undistortion. Did you see any issues in the final calibration results?

closed 2023-07-13 by koide3

yahyahajlaoui, 2023-08-02:
> The previous issue has been resolved, and I am now able to calibrate it. However, I am encountering skewed and distorted images after preprocessing them. Here is an example image
> 
> [image]
> 
> It is worth noting that the input data from the rosbag does not exhibit any distortion or skewing. When choosing the corresponding points for the initial guess, I am provided with a regular(standard), undistorted image. Here is an example:
> 
> [image]

Hello, can you please help me how do you solve the issue, I am facing the same. Thank you

Giannis-Alamanos, 2023-10-13:
How did you solve the issue?

pengpengfei97, 2023-11-28:
> * [✓] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
> * [ ✗] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)
> 
> If the issue is about calibration procedure and results:
> 
> * [✓] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
> * [ ]  I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)
> 
> Environment: OS: ubuntu 20.04 ROS Version: noetic
> 
> I am having an issue while running preprocess command rosrun direct_visual_lidar_calibration preprocess data/ process/ -a -d -v
> 
> Error:
> 
> data_path: /lidar-cam-calib/data/ dst_path : /lidar-cam-calib/process/ input_bags:
> 
> * /lidar-cam-calib/data/_2023-04-13-17-49-57_0.bag
> * /lidar-cam-calib/data/_2023-04-13-17-50-27_1.bag
> * /lidar-cam-calib/data/_2023-04-13-17-50-57_2.bag
> * /lidar-cam-calib/data/_2023-04-13-17-51-28_3.bag
> * /lidar-cam-calib/data/_2023-04-13-17-51-57_4.bag
> * /lidar-cam-calib/data/_2023-04-13-17-52-27_5.bag
>   topics:
> * camera_info: /lidar-cam-calib/data/_2023-04-13-17-52-27_5.bag
>   topics:
> * image      :cam0/image_raw
> * points     :/vlp16_1/velodyne_points
>   intensity_channel: intensity
>   terminate called after throwing an instance of 'rosbag::BagIOException'
>   what():  Error opening file: cam0/image_raw
>   Aborted (core dumped)

Hi, friend. I meet the same issue: `Error opening file: /image`. I read the codes and found it perhaps caused by the function https://github.com/koide3/direct_visual_lidar_calibration/blob/82df9955c2bf43a478690d45f9283493ca01343d/src/preprocess_ros1.cpp#L98C110-L98C110 when passing parameters. This function read the `image_topic` as `bag_filename` and can't get the parameter `image_topic` correctly. But I have no idea how to fix the problem, cloud you please help me how you solved this problem? Thanks!