koide3/direct_visual_lidar_calibration #9: Could not reproduce calibration example with docker

0xd1ma, 2023-04-24:
**Before opening an issue**
If the issue is about build errors:
- [ ] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [ ] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [ ] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [ ] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [ ] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
when repeating the [instruction](https://koide3.github.io/direct_visual_lidar_calibration/example/) with running example using docker for ros1
in console:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. download [dataset](https://koide3.github.io/direct_visual_lidar_calibration/example/)
2. get docker [image](https://koide3.github.io/direct_visual_lidar_calibration/docker/)
3. run
`docker run   -it   --rm   --net host   --gpus all   -e DISPLAY=$DISPLAY   -v $HOME/.Xauthority:/root/.Xauthority   -v /home/robot/calibration/input_bags:/tmp/input_bags   -v /home/robot/calibration/preprocessed:/tmp/preprocessed   koide3/direct_visual_lidar_calibration:noetic   rosrun direct_visual_lidar_calibration preprocess -av   /tmp/input_bags /tmp/preprocessed`

**Expected behavior**
reproduce calibration example.

**Screenshots and sample data**
If applicable, add screenshots and sample data to reproduce your problem.

**Environment:**
 - OS: Ubuntu 22.04
 - ROS version: ROS1 noetic

**Additional context**
Add any other context about the problem here.

0xd1ma, 2023-04-24:
> 

it is just a names
[image]

koide3, 2023-04-25:
This is a known issue caused by ROS2-ROS1 bag conversion. For the provided example data, you need to manually specify the camera intrinsic parameters. Note that the program should work for bags that directly recorded using ROS1.

https://koide3.github.io/direct_visual_lidar_calibration/example/

```
CameraInfo messages in ROS1 bag files exhibit an MD5 checksum that is different from that of the standard sensor_msgs/CameraInfo msg due to ROS2-ROS1 bag conversion issues. Because this affects the automatic camera info extraction, you need to manually specify camera parameters for preprocessing.
```

0xd1ma, 2023-04-25:
My apologies for not reading the instructions carefully

0xd1ma, 2023-04-25:
for my data:
[image]

is it because using compressed image ?

koide3, 2023-04-26:
At this moment, the ROS1 version does not support CompressedImage, but I'll update the code to support it soon.

ashwinjoseph95, 2024-11-18:
Hi @koide3 ,
I am trying to run the calibration on a similar environemnt (ROS1 and Ubuntu 20)
I have setup the docker image provided by you. I was able to do the preprocess step but when it came to next step for manually selcting points I realized there was a lot of lag and also that the docker was not using gpu even though I have provided the flags for it to do so.
My commands are as follows:
```
# Define paths
bag_path=$(realpath /home/knadmin/Ashwin/Targetless_calib/livox_ros1)
preprocessed_path=$(realpath /home/knadmin/Ashwin/Targetless_calib/livox_ros1_preprocessed)

docker run \
  -it \
  --net host \
  --runtime=nvidia --gpus all  \
  -e DISPLAY=$DISPLAY \
  -v $HOME/.Xauthority:/root/.Xauthority \
  -v $bag_path:/tmp/input_bags \
  -v $preprocessed_path:/tmp/preprocessed \
  -v pip_cache:/root/.cache/pip \
  koide3/direct_visual_lidar_calibration:noetic /bin/bash nvidia-smi
```
  but apparenty nvidia-smi is not getting recognized. I have insalled all the nvidia and cuda dependencies in the local system.
  I tried another example here with ubuntu docker iamge it detects the nvidia drivers:
[image]
What should i do to rectify this issue? I understand this might be slightly off topic but I would be really helpful to udnerstand what is going wrong and how I can fix it.

koide3, 2024-11-24:
It seems like the issue stems from the docker environment not properly recognizing GPU for OpenGL. I'm not very familiar with OpenGL-related docker settings, but I think changing the base docker image to [nvidia/opengl](https://hub.docker.com/r/nvidia/opengl) may resolve the problem.