koide3/direct_visual_lidar_calibration #12: I find a bug when using find_matches_superglue.py

Jeasonlee313, 2023-04-27:
I think the bug might be caused for the version of matplotlib
When I was using find_matches_superglue.py to automatic initial guess, there raises a ValueError: "Colormap turbo is not recognized. Possible values are: Accent, Accent_r, blabla..." in line 136 in direct_visual_lidar_calibration/scripts/find_matches_superglue.py.
I use matplotlib 3.1.2, could you like to give me your matplotlib version to check the bug can be solved or not?

koide3, 2023-04-27:
```
$ pip show matplotlib
Name: matplotlib
Version: 3.5.1
```

Jeasonlee313, 2023-04-27:
> ```
> $ pip show matplotlib
> Name: matplotlib
> Version: 3.5.1
> ```

Thank you! I can use find_matches_superglue.py to auto initial guess, but there are still some misalignment between camera and Lidar in the test data livox

Jeasonlee313, 2023-04-27:
[image]

Jeasonlee313, 2023-04-27:
> > ```
> > $ pip show matplotlib
> > Name: matplotlib
> > Version: 3.5.1
> > ```
> 
> Thank you! I can use find_matches_superglue.py to auto initial guess, but there are still some misalignment between camera and Lidar in the test data livox

I'm sorry, I input wrong intrinsic parameters, and now I can get the correct results. I will close this issue. Thanks for your help a lot!

Jeasonlee313, 2023-04-27:
[image]

closed 2023-04-27 by Jeasonlee313