koide3/direct_visual_lidar_calibration #13: Velodyne 16 line radar cannot complete calibration

af-doom, 2023-05-01:
**Before opening an issue**
If the issue is about build errors:
- [1] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [1] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [1] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [1] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [ ] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
 U
[image]
nable to generate dense point clouds during the data processing stage of velodyne16 line LiDAR

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots and sample data**
If applicable, add screenshots and sample data to reproduce your problem.

**Environment:**
 - OS:Ubuntu 20.04
 - ROS version: ROS1 noetic

**Additional context**
Add any other context about the problem here.

Ning030, 2023-06-29:
Have you solved this problem? If so, can you provide a solution? Thanks Thanks!

koide3, 2023-07-05:
Can anybody provide an example rosbag to reproduce the issue?

af-doom, 2023-07-05:
> Have you solved this problem? If so, can you provide a solution? Thanks Thanks!

sorry ,no

Ning030, 2023-07-18:
This is a ROS1 bag that I recorded myself.
[(https://drive.google.com/file/d/1Z_S8l-WU6jQWuUJfvNQVFbGRWMgjAY2d/view?usp=sharing)](url)
 This is a robosence 16 line LiDAR .
Unable to generate dense point clouds during the data processing stage!!!

koide3, 2023-07-19:
Hi @Ning030 ,
Thanks for providing the rosbag. I found that the issue stems from input point clouds containing NaN points. I added a filter to remove invalid points and pushed the updated code to `vlp16` branch, which will be merged into `main` soon.

Ning030, 2023-07-20:
Thank you for your answer. I greatly admire your excellent work！！
I would like to ask, **how to use the automatic initial guess（SuperGlue）**?
 I downloaded the [source code](https://github.com/magicleap/SuperGluePretrainedNetwork), but the calibration effect is not good.
Did you retrain it？

koide3, 2023-07-24:
The accumulation result may not have a sufficient point density required for the automatic initial guess pipeline. Try to increase the  LiDAR data (with up-down movement) or to use the manual initial guess interface.

Ning030, 2023-07-28:
Thank you for your suggestion. 
But I still have **poor calibration results** when **using the data you provided** to run the algorithm. （With automatic initial guess）.
Why is this?

Ning030, 2023-08-02:
Sorry！
 I found out that it was caused by my input of wrong camera intrinsic data. 
Thank you for making this work public.

closed 2023-08-03 by koide3