koide3/direct_visual_lidar_calibration #14: Installation Problem

zza1617, 2023-06-03:
CMake Error at CppUnitLite/CMakeLists.txt:6 (add_library):
  Target "CppUnitLite" links to target "Boost::boost" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at gtsam/CMakeLists.txt:107 (add_library):
  Target "gtsam" links to target "Boost::serialization" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at gtsam/CMakeLists.txt:107 (add_library):
  Target "gtsam" links to target "Boost::system" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at gtsam/CMakeLists.txt:107 (add_library):
  Target "gtsam" links to target "Boost::filesystem" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
how to deal with it?

koide3, 2023-06-05:
You may need to do `sudo apt install libboost-all-dev`

zza1617, 2023-06-05:
I think I have installed libboost-all-dev. 
/usr/lib/x86_64-linux-gnu includes boost_ ...

koide3, 2023-06-07:
Did you get the error during the build of GTSAM?

zza1617, 2023-06-07:
I have resolved the previous issue by replacing errors like "Boost::boost" with "${Boost_LIBRARY}". The problem has been resolved.

zza1617, 2023-06-10:
When I Install Iridescence for visualization,
I meet this error: 
/home/zza/iridescence/src/guik/viewer/info_window.cpp:11:10: fatal error: boost/process/child.hpp: No such file or directory
11 | #include <boost/process/child.hpp>
| ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
How to resolve this issue?
look forward to your reply.
thanks

zza1617, 2023-06-11:
My system is ubuntu16.04, boost1.58 @koide3

koide3, 2023-06-12:
Sorry, we do not support environments older than Ubuntu 20.04. But you can still use this package without installation by using docker images.

closed 2023-07-05 by koide3

BestVanRome, 2024-07-08:
@koide3 your docker-files seems to be missing the  `Iridescence_LIBRARIES`. (Both: noetic and humble).
Got errors on trying to build both.

koide3, 2024-07-09:
@BestVanRome 
Thanks for reporting. The error should be fixed by https://github.com/koide3/direct_visual_lidar_calibration/pull/96 .

BestVanRome, 2024-07-09:
> @koide3 your docker-files seems to be missing the `Iridescence_LIBRARIES`. (Both: noetic and humble). Got errors on trying to build both.

@koide3 do you want to update your docker-files as well?

koide3, 2024-07-09:
@BestVanRome 
No, it's not necessary. The error was fixed by the modification made on FindIridescence.cmake.