koide3/direct_visual_lidar_calibration #15: Some questions about input data in paper

Karry11, 2023-06-06:
hello, I am a beginner in the field of radar camera calibration. After I read yours paper and code, I think it is intersting and feel excited. But I have some question about the process. 
  In your paper, the input of preprocessing are Lidar data and Camera images, but when I read code and livox data, the input data just like *.db3. So, Is the image derived from point cloud data?maybe you use other ways to generate the images? If I have images and lidar data, can I skip preprocessing? Thanks a lot.

koide3, 2023-06-07:
The db3 file contains LiDAR and image data in the ROS2 bag format. You can confirm the contents in a rosbag as follows:

```
$ ros2 bag info livox/rosbag2_2023_03_09-13_42_46/
# Files:             rosbag2_2023_03_09-13_42_46_0.db3
# Bag size:          582.9 MiB
# Storage id:        sqlite3
# Duration:          15.650s
# Start:             Mar  9 2023 13:42:46.665 (1678336966.665)
# End:               Mar  9 2023 13:43:02.316 (1678336982.316)
# Messages:          2972
# Topic information: Topic: /livox/points | Type: sensor_msgs/msg/PointCloud2 | Count: 157 | Serialization Format: cdr
#                    Topic: /livox/imu | Type: sensor_msgs/msg/Imu | Count: 2597 | Serialization Format: cdr
#                    Topic: /livox/lidar | Type: livox_interfaces/msg/CustomMsg | Count: 157 | Serialization Format: cdr
#                    Topic: /image | Type: sensor_msgs/msg/Image | Count: 30 | Serialization Format: cdr
#                    Topic: /camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 31 | Serialization Format: cdr
```

If you have images and corresponding point clouds, you can skip the preprocessing as in the following example. Note that this procedure may not be well tested compared to the standard procedures using rosbags.

https://koide3.github.io/direct_visual_lidar_calibration/example_map/

Karry11, 2023-06-08:
Thanks a lot, I have figure out the problem. By the way, is there the code for testing T_lidar_camera params and get  Trans or Rot?

koide3, 2023-06-08:
In the calibration result file `calib.json`, `T_lidar_camera` represents `[tx, ty, tz, qx, qy, qz, qw]`. If you need to convert the quaternion to another rotation representation use some tools like [this](https://www.andre-gaschler.com/rotationconverter/).

closed 2023-06-12 by Karry11