koide3/direct_visual_lidar_calibration #16: Installation error

andy690, 2023-06-07:
**Before opening an issue**
If the issue is about build errors:
- [ ] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)

**Describe the bug**
Error occurred when install iridescence

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots and sample data**
/home/agilex/iridescence/thirdparty/imgui/backends/imgui_impl_glfw.cpp: In function ‘void ImGui_ImplGlfw_UpdateMonitors()’:
/home/agilex/iridescence/thirdparty/imgui/backends/imgui_impl_glfw.cpp:776:9: error: ‘glfwGetMonitorWorkarea’ was not declared in this scope
         glfwGetMonitorWorkarea(glfw_monitors[n], &x, &y, &w, &h);
         ^~~~~~~~~~~~~~~~~~~~~~
/home/agilex/iridescence/thirdparty/imgui/backends/imgui_impl_glfw.cpp:776:9: note: suggested alternative: ‘glfwGetMonitorName’
         glfwGetMonitorWorkarea(glfw_monitors[n], &x, &y, &w, &h);

**Environment:**
 - Ubuntu 18.04
 - ROS 1

koide3, 2023-06-08:
Possibly, the version of GLFW in Ubuntu 18.04 is too old. Try to upgrade it or use docker images to do calibration without installation.

andy690, 2023-06-08:
My GLFW version is 3.3.2, will it be able to support the project?

koide3, 2023-06-08:
We'll not support environments older than Ubuntu 20.04. Please consider using docker images for older environments.

closed 2023-07-05 by koide3

cwha0212, 2023-09-13:
I use ubuntu 20.04 and ROS1 but, I got same error. how can i solve this?