koide3/direct_visual_lidar_calibration #22: Another error from using docker environment

Ning030, 2023-06-20:
There is an error when i run：
ros2 run direct_visual_lidar_calibration find_matches_superglue.py ouster_preprocessed

ModuleNotFoundError: No module named 'torch'

james-yoo, 2023-06-20:
@Ning030 
You should install pytorch module as follow:
```
pip install torchvision
```

closed 2023-07-05 by koide3

BestVanRome, 2024-07-05:
For convenience and more automatism ;-), would be nice just to add it to your dockerfile or hub ;-).
@koide3 

(Great Project so far btw 👍🏽 )

> @Ning030 You should install pytorch module as follow:
> 
> ```
> pip install torchvision
> ```

koide3, 2024-07-05:
@BestVanRome 
We intentionally avoided including SuperGlue-related things in the docker image due to SuperGlue's strict license. Instead, we provide `Dockerfile_with_superglue` so that the user can build the image with SuperGlue by themselves.

https://github.com/koide3/direct_visual_lidar_calibration/blob/main/docker/humble/Dockerfile_with_superglue
https://github.com/koide3/direct_visual_lidar_calibration/issues/83

BestVanRome, 2024-07-05:
Unfortunately I still am not able to build it: 

```bash
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
4.115   Could NOT find Iridescence (missing: Iridescence_LIBRARIES)
4.115 Call Stack (most recent call first):
4.115   /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
4.115   direct_visual_lidar_calibration/cmake/FindIridescence.cmake:16 (find_package_handle_standard_args)
4.115   direct_visual_lidar_calibration/CMakeLists.txt:28 (find_package)
```
and

```bash
UndefinedVar: Usage of undefined variable '$PYTHONPATH' (line 49)
```
:-(