koide3/direct_visual_lidar_calibration #23: Lidar-Camera Calibration for fisheye lense

gitissues79, 2023-06-22:
**Before opening an issue**
If the issue is about build errors:
- [✓ ] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [ ] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [ ✓] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [✓ ] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [ ✓] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**

I am using a fisheye lens for lidar camera calibration, but I'm encountering issues with the calibration process. Specifically, I am using rectified fisheye images for the camera calibration process. The distortion model I'm using, "equidistant," is throwing an error. I referred to the official pages "http://wiki.ros.org/camera_calibration" and "http://docs.ros.org/en/melodic/api/sensor_msgs/html/namespacesensor__msgs_1_1distortion__models.html#a51077dbd502bdea20f46493798ca807b" for information. According to these resources, for the fisheye lens in the Noetic version, the distortion model should be "equidistant."

I am able to preprocess it, I am facing an issue in Initial guess manual, throwing an error " error: unknown camera model equidistant"
command: rosrun direct_visual_lidar_calibration initial_guess_manual /media/user/fisheye/fish_test_2/process/ 

Upon reviewing the code, I noticed that the distortion model for the fisheye lens is specified as "FISHEYE." I attempted to replace this model with "equidistant," but it did not work as expected. Although I am able to obtain the image for choosing corresponding points, the target lidar points are missing. in the terminal it is showing Camera fov as 0 deg.

[image]

**Environment:**
 - OS: [Ubuntu 20.04]
 - ROS version: [ROS1 noetic]

Jeasonlee313, 2023-06-30:
I also encountered this problem. I use the fisheye camera intrinsic parameters which are calibrated by ros camera calibration tool box, and input these parameter by using --camera_intrinsics and --camera_distortion_coeffs in node "preprocess". And, during I use node "initial_guess_manual" to get a rough extrinsic parameter, I also get "camera_fov:0[deg]", and get an error about rotation matrix:
R is not orthogonal:
 0 0 0
0 0 0
0 0 0

I check the "fisheye.hpp" in "include/camera", found the fisheye projection_model is pinhole but distortion_model is fisheye. In opencv, for fisheye lens, there is a specifical module fisheye::projectPoints to project points to pixels, maybe the projection_model should also be fisheye? This is just my guess, looking forward to resolving this bug!

koide3, 2023-07-05:
@gitissues79 
I think this problem can easily be resolved by slightly modifying `create_camera.cpp` so that it accepts `equidistant` as `fisheye`. Can you provide a rosbag so that I can reproduce the issue and test the update?

koide3, 2023-07-05:
@Jeasonlee313 
Indeed, the fisheye model in OpenCV is the combination of the pinhole projection and the equidistant lens distortion, and thus it cannot handle a camera with an FoV larger than 180deg (I wrote a test code and confirmed that our implementation gives the exactly the same results as those of `cv::fisheye::projectPoints`). If you need to use an ultra wide FoV camera, consider using [`omnidir` camera model](https://rpg.ifi.uzh.ch/docs/omnidirectional_camera.pdf) instead.

Jeasonlee313, 2023-07-05:
> @Jeasonlee313 Indeed, the fisheye model in OpenCV is the combination of the pinhole projection and the equidistant lens distortion, and thus it cannot handle a camera with an FoV larger than 180deg [...]

@koide3 Thank you for your reply. Actually, I calibrate my fisheye camera use opencv calibration modules. So, I have the intrinsic parameters contains  "fx, fy, cx, cy and k1, k2, k3, k4". I use these parameters to initial extrinsic parameters manually, but, when I want to solve the extrinsic parameters, I got the error "R is not orthogonal", and I found the "camera_fov:0[deg]", it is a weird phenomenon. Maybe my question is similarity like @gitissues79.

Jeasonlee313, 2023-07-05:
> @Jeasonlee313 Indeed, the fisheye model in OpenCV is the combination of the pinhole projection and the equidistant lens distortion, and thus it cannot handle a camera with an FoV larger than 180deg [...]

@koide3  Here is my rosbag recorded with ros1 noetic https://github.com/Jeasonlee313/public_data/blob/main/2023-06-29-19-30-32.bag, and the camera intrinsic parameter is --camera_intrinsics 511.244537,511.573395,986.686218,562.944397 --camera_distortion_coeffs 0.117343001,-0.0315180011,0.000775000022,0.000146000006, thank you very much, I hope the data can help you to test the update.

koide3, 2023-07-05:
Hmm, in my environment, the program estimated the camera FoV as `174.123[deg]` that looks fine. Can you check if the following options work?

```bash
--camera_model fisheye --camera_intrinsics 511.244537,511.573395,986.686218,562.944397 --camera_distortion_coeffs 0.117343001,-0.0315180011,0.000775000022,0.000146000006
```

Jeasonlee313, 2023-07-05:
> Hmm, in my environment, the program estimated the camera FoV as `174.123[deg]` that looks fine. Can you check if the following options work?
> 
> ```shell
> --camera_model fisheye --camera_intrinsics 511.244537,511.573395,986.686218,562.944397 --camera_distortion_coeffs 0.117343001,-0.0315180011,0.000775000022,0.000146000006
> ```

@koide3 
Well, in my environment, after I prepare the data by using preprocess, I got the calib.json like this:
{
  "camera": {
    "camera_model": "fisheye",
    "distortion_coeffs": [
      0.117343001,
      -0.0315180011,
      0.000775000022,
      0.000146000006
    ],
    "intrinsics": [
      511.244537,
      511.573395,
      986.686218,
      562.944397
    ]
  },
  "meta": {
    "bag_names": [
      "2023-06-29-19-30-32.bag"
    ],
    "camera_info_topic": "",
    "data_path": "/home/li/tt_ws/bag",
    "image_topic": "/front_right",
    "intensity_channel": "intensity",
    "points_topic": "/pcl_output"
  }
}

And then, I use the initial_guess_manual to estimate an initial extrinsic parameter, here is the log in the terminal:
loading data/fish/front/2023-06-29-19-30-32.bag.(png|ply)
warning: face properties dont exist!!
camera_fov:0[deg]

and for visualization, few lidar points are highlighted.
[image]

It there any changes on your local code branch?

koide3, 2023-07-05:
I found that the FoV estimation sometimes returns NaN for the fisheye model. I pushed a fix to `fisheye` branch, and it'll be merged into `main` branch soon. Can you see if this update fixes the issue?

Jeasonlee313, 2023-07-06:
> I found that the FoV estimation sometimes returns NaN for the fisheye model. I pushed a fix to `fisheye` branch, and it'll be merged into `main` branch soon. Can you see if this update fixes the issue?

@koide3 
sorry, I have checkout to the branch "fisheye", and try to catkin_make the project again. But, I got several errors during the making project: 
"fatal error: sophus/se3.hpp: No such file or directory" and "fatal error: nlohmann/json.hpp: No such file or directory"
Is there any additional dependences? Except sophus, is there any other dependences you did not write in README?

Jeasonlee313, 2023-07-06:
> I found that the FoV estimation sometimes returns NaN for the fisheye model. I pushed a fix to `fisheye` branch, and it'll be merged into `main` branch soon. Can you see if this update fixes the issue?

@koide3 
Hi, I have installed Sophus and json. And, I use your fisheye branch to remake all calibration project. It does works! Thank you for you help, and I think you can merge the fisheye branch to main branch, and also you need to modify the README with adding Sophus and json in Dependences.

koide3, 2023-07-10:
Thanks for reporting that. Actually, the installation of Sophus and json is not necessary as they are bundled in ```thirdparty``` directory but you need to clone them as submodules as follows: 

```bash
git clone https://github.com/koide3/iridescence --recursive
```

closed 2023-07-13 by koide3

xl232lx, 2025-06-30:
Hello, the hardware I use is a wide-angle camera (with a field of view of approximately 160 degrees) and livox mid360 lidar. For the --camera_model parameter, I use fisheye and euqidistant. The external parameter results obtained from them all have very poor coloring effects. May I ask what is usually the reason for this?