koide3/direct_visual_lidar_calibration #25: How to specify camera parameters for preprocessing

774880930, 2023-07-04:
**Before opening an issue**

If the issue is about calibration procedure and results:
- [√] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [√] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

**Describe the bug**
I used ROS1. When I was preprocessing, I saw the notes you provided and needed to specify the parameters of the camera in advance, but when I tried to specify the parameters, I got the following results 

[image]
[image]
[image]

What command should I type to do it？

**Environment:**
 - OS: [e.g. Ubuntu 20.04]
 - ROS version: [e.g., ROS1 noetic]

koide3, 2023-07-05:
You need to specify the camera distortion params in addition to the intrinsic params as follows:

```cpp
 rosrun direct_visual_lidar_calibration preprocess -av \
  --camera_model plumb_bob \
  --camera_intrinsic 1452.711762456289,1455.877531619469,1265.25895179213,1045.818593664107 \
  --camera_distortion_coeffs -0.04203564850455424,0.0873170980751213,0.002386381727224478,0.005629700706305988,-0.04251149335870252 \
  /tmp/input_bags /tmp/preprocessed
```

koide3, 2023-07-05:
Possibly, you forgot the backslash at the end of each line?

774880930, 2023-07-05:
I've tried that,but it doen't work
[image]
[image]

774880930, 2023-07-05:
Oh, Yes, I forgot "/" at the end of each line,I thought it meant "or",this is really my misunderstanding!!
Thanks a lot!!!

774880930, 2023-07-05:
One more question, if I want to calibrate a line laser and camera, and the line laser changes different positions through the galvanometer, can this package be applied?

koide3, 2023-07-06:
If you mean a 2D line laser, this package cannot be applied to it. It requires a 3D range sensor.

closed 2023-07-10 by koide3

Chatoyant19, 2024-01-30:
> You need to specify the camera distortion params in addition to the intrinsic params as follows:
> 
> ```c++
>  rosrun direct_visual_lidar_calibration preprocess -av \
>   --camera_model plumb_bob \
>   --camera_intrinsic 1452.711762456289,1455.877531619469,1265.25895179213,1045.818593664107 \
>   --camera_distortion_coeffs -0.04203564850455424,0.0873170980751213,0.002386381727224478,0.005629700706305988,-0.04251149335870252 \
>   /tmp/input_bags /tmp/preprocessed
> ```

For the two sets of data you provided, livox and ouster, how do I get their camera intrinsics and _distortion_coeffs. I can't seem to parse this information from '/camera_info', because the version of the topic does not match.

G4419, 2024-03-20:
@Chatoyant19 you can use plotjuggler to see the data.