koide3/direct_visual_lidar_calibration #29: Can this methodology

liuXinGangChina, 2023-07-06:
Thanks for your great work!
I have already test your dataset with the docker image(ros2) you provide.
However, when i try with my own data, the automatic point integration fails. Maybe there is some paramters that i can tune?
Here is my data's description:
Lidar , solid-state lidar ,120 degree fov, repeative lidar
Rather than move the lidar up and down for a while, i move the lidar strait Forward for a few meters slowly( because the lidar is installed on a vehicle).
The point intergration node's output trajectory looks that the lidar did not moving at all.

liuXinGangChina, 2023-07-07:
Additional info about the lidar i used in this test
lidar spec [https://www.robosense.ai/en/RS-LiDAR-M1](url)
lidar pointcloud demonstration [https://cdn.robosense.cn/4%20%E7%AC%AC%E5%9B%9B%E7%89%88%EF%BC%8C%E7%82%B9%E4%BA%91%E8%B4%A8%E9%87%8F%20%E4%B8%AD%E8%8B%B1%E6%96%87%20%E5%90%8C%E4%B8%80%E6%96%87%E4%BB%B6.mp4](url)

koide3, 2023-07-10:
Did you enable the dynamic points integration with ```-d``` option? If possible, providing an example rosbag would be a help to dig into the problem.

liuXinGangChina, 2023-07-10:
Thanks for your reply， i use -adv

koide3, 2023-07-20:
Possibly the issue stems from input point clouds containing NaN points https://github.com/koide3/direct_visual_lidar_calibration/issues/13. I just merged updated code to fix the issue https://github.com/koide3/direct_visual_lidar_calibration/pull/36.

liuXinGangChina, 2023-07-24:
> Possibly the issue stems from input point clouds containing NaN points #13. I just merged updated code to fix the issue #36.

Thanks for your effort , i will give it a try and feed back to you.

closed 2023-07-24 by liuXinGangChina