koide3/direct_visual_lidar_calibration #44: "Segmentation fault" in Preprocessing

Benson722, 2023-08-21:
**Describe the bug**
When I try to preprocess the data in **livox_ros1.tar.gz**, it always goes wrong. I followed the instructions every step of the way, but I don't know why it happens. 

By the way, the highest CPU usage rate is 59.1%.

**Screenshots and sample data**
```
pi@ubuntu:~/catkin_ws$ rosrun direct_visual_lidar_calibration preprocess /mnt/hgfs/DATA_SENSOR/livox livox_preprocessed -av
data_path: /mnt/hgfs/DATA_SENSOR/livox
dst_path : livox_preprocessed
input_bags:
- /mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_42_46.bag
- /mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_44_10.bag
- /mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_44_54.bag
- /mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_46_10.bag
- /mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_46_54.bag
topics in /mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_42_46.bag:
- /livox/points : sensor_msgs/PointCloud2
- /livox/imu : sensor_msgs/Imu
- /image : sensor_msgs/Image
- /camera_info : sensor_msgs/CameraInfo
selected topics:
- camera_info: /camera_info
- image      : /image
- points     : /livox/points
intensity_channel: intensity
try to get the camera model automatically
warning: bag does not contain topic
       : bag_filename=/mnt/hgfs/DATA_SENSOR/livox/rosbag2_2023_03_09-13_42_46.bag
       : topic=/camera_info
Segmentation fault (core dumped)
```

**Environment:**
 - OS: Ubuntu 20.04
 - ROS version: ROS1 noetic

koide3, 2023-09-13:
For the ROS1 example bag files, you need to manually specify the camera intrinsic parameters:

https://koide3.github.io/direct_visual_lidar_calibration/example/

```
CameraInfo messages in the example ROS1 bag files exhibit an MD5 checksum that is different from that of the standard sensor_msgs/CameraInfo msg due to ROS2-ROS1 bag conversion issues. Because this affects the automatic camera info extraction, you need to manually specify camera parameters for preprocessing.
```

Benson722, 2023-09-26:
Thank you for your help! I'm sorry for replying so late. The problem is solved. :)

closed 2023-09-26 by Benson722

kzhao94, 2024-07-08:
> Thank you for your help! I'm sorry for replying so late. The problem is solved. :)

I am currently struggling with this bug. Could you please share how you resolved it?

kzhao94, 2024-07-08:
https://github.com/koide3/direct_visual_lidar_calibration/issues/44#issuecomment-1734931164

I am currently struggling with this bug. Could you please share how you resolved it?

keepermzw, 2024-10-14:
Could you tell me how to solve the problem?please  I have the same problem 
[image]

leoxiang66, 2025-01-02:
> > Could you tell me how to solve the problem?please I have the same problem ![Screenshot from 2024-10-14 17-51-31](https://private-user-images.githubusercontent.com/165362480/376190771-6e91e505-be4e-49b6-9fb3-24ec63735d97.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE5OTI1NzQsIm5iZiI6MTczMTk5MjI3NCwicGF0aCI6Ii8xNjUzNjI0ODAvMzc2MTkwNzcxLTZlOTFlNTA1LWJlNGUtNDliNi05ZmIzLTI0ZWM2MzczNWQ5Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExOVQwNDU3NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYjUzYTg1M2E5NDFlMzBkNDAxOGM3ZjYzNGEwYmY1ZjhmZjhlN2M4MTkxZTBlODU4MzdhMjg5ZmM1MGViNjQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.MKz5aNV8DT52etuWPuuhAFmT1LMe-mAQktgd8sUQt80)
> 
> You need to specify --camera_model, --camera_dist_coeffs and --camera_instrinsics in your command line, and those parameters are specified in programs.md for example data

Hi Theon, could you give a concrete example command? I am encountering the same problem and I really need some help. really appreciate it!