koide3/direct_visual_lidar_calibration #45: Request for the "pseudo" ground truth from livox_ros1 dataset and configuration for Fisheye Cameras

Es1erda, 2023-08-27:
I'd like to express my sincere gratitude for open-sourcing such an outstanding piece of work. This is, by far, the most user-friendly and efficient calibration tool I have ever used!

While testing on your provided dataset, I found that the results were exceptionally accurate. I noticed in your paper that you mentioned using the "pseudo" ground truth as a reference. I'm keen to see how the results from your algorithm align with this "pseudo" ground truth to understand its precision further.

Additionally, I am interested in using a fisheye camera. Could you guide me on how to adjust the configuration parameters in the files to accommodate this type of camera?

If you could share this information and the data, I would be immensely grateful. 

Thank you once again for your remarkable contribution to the community.

koide3, 2023-09-01:
Thank you for your request. At this moment, I'm fully occupied with some works and cannot have time to make the dataset public. But, I'll consider releasing it once I become free.

The implemented fisheye camera model is identical to the one in OpenCV, and you need to provide distortion coefficients (k1, k2, k3, k4) in addition to camera intrinsics (fx, fy, cx, cy). The calibration process itself is the same as that for the standard camera model.

https://docs.opencv.org/3.4/db/d58/group__calib3d__fisheye.html
https://github.com/koide3/direct_visual_lidar_calibration/blob/main/include/camera/fisheye.hpp

Es1erda, 2023-09-01:
Thank you for taking the time to respond to me amidst your busy schedule! I have already learned how to calibrate fisheye cameras using your open-source toolkit.

Perhaps I wasn't clear enough in my previous message. What I meant to ask is, could you provide the 'pseudo' ground truth of the extrinsic parameters for the devices in your livox_ros1.tar.gz (ROS1) dataset? I would like to use it as a reference.

Once again, thank you very much for your kind assistance!

koide3, 2023-09-13:
Ah, I got it. It think I can provide the LiDAR-IMU transformation data tomorrow.

koide3, 2023-09-14:
Sorry, I just noticed that the camera attached to the Livox LiDAR was exchanged after the quantitative evaluation we conducted in the paper, and the pseudo groundtruth is not available for the configuration of the example data.

Es1erda, 2023-09-15:
Thank you for your response. It's alright that the pseudo groundtruth is not available. I apologize for taking up your valuable time. Wishing you success in your future research endeavors and a happy life. I look forward to seeing your future achievements！

closed 2023-09-15 by Es1erda