koide3/direct_visual_lidar_calibration #46: Coordinate frame between Lidar and Camera

monamourvert, 2023-08-28:
Hi, thanks for your great work! I'm still  a newbie for using lidar and camera. So, I just want to ask about the coordinate frame of both sensor. Since my **LiDAR (ouster) Coordinate System**:

- X-axis: Forward (towards the direction the vehicle moves)
- Y-axis: Left
- Z-axis: Upwards

**Camera Coordinate System (Azure Kinect, OpenCV-based)**:

- X-axis: Right (pointing to the right of the image)
- Y-axis: Down (pointing downwards in the image)
- Z-axis: Forward (out of the camera lens)

Then does the calibration result, the **T_lidar_camera**  already overcome this change? Or should I flip it by myself? Thanks!

koide3, 2023-09-01:
As long as both the sensors obey the right-hand system convention, the difference of LiDAR-camera coordinate systems doesn't matter. `T_lidar_camera` is a transformation that brings a points in the camera frame (`pt_camera`) to the LiDAR frame (`pt_lidar = T_lidar_camera * pt_camera`).

monamourvert, 2023-09-01:
Thanks! I finally figure it out!

closed 2023-09-01 by monamourvert

l1997i, 2024-06-05:
> As long as both the sensors obey the right-hand system convention, the difference of LiDAR-camera coordinate systems doesn't matter. `T_lidar_camera` is a transformation that brings a points in the camera frame (`pt_camera`) to the LiDAR frame (`pt_lidar = T_lidar_camera * pt_camera`).

The definition provided for `T_lidar_camera` seems to be reversed in terms of **common notation** used in coordinate transformations. Typically, the transformation matrix `T_lidar_camera` is understood to transform points from the LiDAR frame to the camera frame. However, the description you provided suggests the opposite—that it transforms points from the camera frame to the LiDAR frame. 

It is more conventionally and correctly named `T_camera_to_lidar`.

koide3, 2024-06-06:
`T_lidar_camera` is not `camera_to_lidar`. It's following the commonly used systematic coordinate frame notation style (e.g., In [GTSAM](https://gtsam.org/gtsam.org/2020/06/28/gtsam-conventions.html), `wTc` is a transformation that transforms a point from frame `c` to `w`, thus `Pw = wTc * Pc`. Our notation just shifts `T` to the left `wTc => T_w_c`).

This style allows easily chain transformations (consecutive frames must be canceled out: `T_w_c * T_c_l * T_l_i = T_w_i`), and is less prone to bugs compared to the style like `camera_to_lidar` that can be counter-intuitive when chaining transformations (`camera_to_world * lidar_to_camera * imu_to_lidar = imu_to_world`).

l1997i, 2024-06-11:
> `T_lidar_camera` is not `lidar_to_camera`. It's following the commonly used systematic coordinate frame notation style (e.g., In [GTSAM](https://gtsam.org/gtsam.org/2020/06/28/gtsam-conventions.html), `wTc` is a transformation that transforms a point from frame `c` to `w`, thus `Pw = wTc * Pc`. Our notation just shifts `T` to the left `wTc => T_w_c`).
> 
> This style allows easily chain transformations (consecutive frames must be canceled out: `T_w_c * T_c_l * T_l_i = T_w_i`), and is less prone to bugs compared to the style like `camera_to_lidar` that can be counter-intuitive when chaining transformations (`camera_to_world * lidar_to_camera * imu_to_lidar = imu_to_world`).

Thank you for your explanations. I found it very useful. By the way, this calibration research is really amazing and it's quite useful to my dataset.

BloodLemonS, 2025-04-24:
Your work is very meaningful and I am also very interested in it, but I still have some questions about the coordinate changes mentioned in the issues. How to represent the transformation relationship of coordinates after converting the calculated quaternion into Euler angles? For example, how is the coordinate system of the outser radar and camera aligned based on the calculated results?