koide3/direct_visual_lidar_calibration #50: An error accured when try to calibrate kitti bag.

guyueyanfeng, 2023-09-13:
If the issue is about calibration procedure and results:
- [X] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [X] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

**Describe the bug**
This work is quite wonderful which makes me want to validate in kitti dataset. However, an error accured when I run the prepropress command. 

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. download kitti dataset: https://drive.google.com/drive/folders/1uFpFjxIBtGPnJbl5NnbUPtcEyx-cpufU?usp=drive_link
     I have use [kitti2bag]() to transform the kitti data into a rosbag.
2. then manually calibrate the bag 
this is the command I used:
'''
rosrun direct_visual_lidar_calibration preprocess kitti_bag kitti_bag_preprocessed \
--image_topic /kitti/camera_color_left/image_raw \
--points_topic /kitti/velo/pointcloud \
--camera_model plumb_bob \
--camera_intrinsics 767.5834,768.4421,200.9368,130.4244 \
--camera_distortion_coeffs -0.0059,-2.4245e-04,-0.4068,0.5534,-8.4002

'''
3. another problem, if I want to calibrate based on a pair of lidar point cloud and image, what should I do. I found that this code need to record at least 2 frames of lidar and camera message for calibrate.

koide3, 2023-09-13:
Possibly points msgs in your bag do not have intensity channels as the error on the console states. Can you check `fields` in your point msgs contain a field named `intensity`?

guyueyanfeng, 2023-09-14:
Oh, thanks. I just realized that the MI could not calculate if the intensity channel lost. What should I do to only obtain the initial parameters without intensity channel?

koide3, 2023-09-14:
Possibly you can feed dummy intensity data and use the manual initial guess UI to align the image with the point cloud by hand.