koide3/direct_visual_lidar_calibration #51: Segmentation fault ouster bag

Jufromerosa, 2023-09-15:
When I try to preprocess the data  ouster for the Ouster-camera calibration it goes wrong. I followed the instructions (https://koide3.github.io/direct_visual_lidar_calibration/example/) step by step, but i get this error when i tried to do the preprocess : 
```
ros2 run direct_visual_lidar_calibration preprocess ouster ouster_preprocessed -adv
data_path: ouster
dst_path : ouster_preprocessed
[INFO] [1694810226.604455129] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_26_51/rosbag2_2023_03_28-16_26_51_0.db3' for READ_ONLY.
[INFO] [1694810226.609745481] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
input_bags:
- ouster/rosbag2_2023_03_28-16_26_51
- ouster/rosbag2_2023_03_28-16_25_54
[INFO] [1694810226.615058493] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
topics in ouster/rosbag2_2023_03_28-16_25_54:
- /camera_info : sensor_msgs/msg/CameraInfo
- /image : sensor_msgs/msg/Image
- /points : sensor_msgs/msg/PointCloud2
selected topics:
- camera_info: /camera_info
- image      : /image
- points     : /points
[INFO] [1694810226.620237488] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
intensity_channel: reflectivity
[INFO] [1694810226.664490737] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
try to get the camera model automatically
[INFO] [1694810226.755682896] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
camera_model: plumb_bob
image_size  : 2448 2048
intrinsics  : 1454.66 1455.21 1229.26 1010.66
dist_coeffs :   -0.0507013     0.111236 -0.000881838  0.000141199   -0.0588915
processing images and points (num_threads_per_bag=12)
start processing ouster/rosbag2_2023_03_28-16_25_54
[INFO] [1694810226.760323450] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
[INFO] [1694810226.869450709] [rosbag2_storage]: Opened database 'ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
[ros2run]: Segmentation fault
```

**Environment:**
 - OS: Ubuntu 22.04 
 - ROS version: ROS2 Humble

koide3, 2023-09-25:
Can you see if the error persists with the docker images (https://koide3.github.io/direct_visual_lidar_calibration/docker/)?
Inconsistent build options of dependent libraries (e.g., march=native flags) may cause this kind of segfaults.

Chatoyant19, 2024-01-30:
> 2448 2048

Hi, how did you get the camera intrinsics?