koide3/direct_visual_lidar_calibration #53: catkin_make error

DoongLi, 2023-09-19:
My environment：Ubuntu20.04，ROS1，gtsam4.0.2，ceres1.14.0；When I used catkin_make, I encountered the following errors:

Base path: /home/doongli/calib_ws
Source space: /home/doongli/calib_ws/src
Build space: /home/doongli/calib_ws/build
Devel space: /home/doongli/calib_ws/devel
Install space: /home/doongli/calib_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/doongli/calib_ws/build"
####
####
#### Running command: "make -j36 -l36" in "/home/doongli/calib_ws/build"
####
[  3%] Building CXX object direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/common/estimate_pose.cpp.o
[  6%] Building CXX object direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp.o
[  9%] Building CXX object direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/calib/visual_camera_calibration.cpp.o
In file included from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_pose.cpp:9:
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/thirdparty/Sophus/sophus/ceres_manifold.hpp:3:10: fatal error: ceres/manifold.h: No such file or directory
    3 | #include <ceres/manifold.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/build.make:154: direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/common/estimate_pose.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/calib/visual_camera_calibration.cpp:10:10: fatal error: ceres/autodiff_first_order_function.h: No such file or directory
   10 | #include <ceres/autodiff_first_order_function.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/build.make:284: direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/calib/visual_camera_calibration.cpp.o] Error 1
In file included from /usr/local/include/gtsam/base/types.h:31,
                 from /usr/local/include/gtsam/global_includes.h:22,
                 from /usr/local/include/gtsam/base/Vector.h:27,
                 from /usr/local/include/gtsam/base/Matrix.h:26,
                 from /usr/local/include/gtsam/base/Manifold.h:22,
                 from /usr/local/include/gtsam/geometry/BearingRange.h:21,
                 from /usr/local/include/gtsam/geometry/Pose3.h:22,
                 from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/include/vlcal/preprocess/dynamic_point_cloud_integrator.hpp:8,
                 from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:1:
/usr/include/tbb/task_scheduler_init.h:21:154: note: #pragma message: TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                                          ^
In file included from /usr/local/include/gtsam/geometry/Unit3.h:36,
                 from /usr/local/include/gtsam/geometry/Rot3.h:25,
                 from /usr/local/include/gtsam/geometry/Pose3.h:24,
                 from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/include/vlcal/preprocess/dynamic_point_cloud_integrator.hpp:8,
                 from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:1:
/usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
In file included from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/include/vlcal/common/integrated_ct_gicp_factor.hpp:61,
                 from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:11:
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/include/vlcal/common/integrated_ct_gicp_factor_impl.hpp: In member function ‘virtual boost::shared_ptr<gtsam::GaussianFactor> vlcal::IntegratedCT_GICPFactor_<TargetFrame, SourceFrame>::linearize(const gtsam::Values&) const’:
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/include/vlcal/common/integrated_ct_gicp_factor_impl.hpp:104:69: error: ‘Hat’ is not a member of ‘gtsam::SO3’
  104 |     H_transed_pose.block<3, 3>(0, 0) = pose.linear() * -gtsam::SO3::Hat(source_pt.template head<3>());
      |                                                                     ^~~
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp: In member function ‘virtual void vlcal::DynamicPointCloudIntegrator::insert_points(const ConstPtr&)’:
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:91:31: error: ‘PriorFactor’ is not a member of ‘gtsam’
   91 |   graph.emplace_shared<gtsam::PriorFactor<gtsam::Pose3>>(0, gtsam::Pose3(pred_T_odom_lidar_begin), gtsam::noiseModel::Isotropic::Precision(6, 1e3));
      |                               ^~~~~~~~~~~
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:91:9: error: parse error in template argument list
   91 |   graph.emplace_shared<gtsam::PriorFactor<gtsam::Pose3>>(0, gtsam::Pose3(pred_T_odom_lidar_begin), gtsam::noiseModel::Isotropic::Precision(6, 1e3));
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:91:55: error: no match for ‘operator>’ (operand types are ‘<unresolved overloaded function type>’ and ‘gtsam::noiseModel::Isotropic::shared_ptr’ {aka ‘boost::shared_ptr<gtsam::noiseModel::Isotropic>’})
   91 |   graph.emplace_shared<gtsam::PriorFactor<gtsam::Pose3>>(0, gtsam::Pose3(pred_T_odom_lidar_begin), gtsam::noiseModel::Isotropic::Precision(6, 1e3));
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |                                                                                        |
      |         <unresolved overloaded function type>                                                    gtsam::noiseModel::Isotropic::shared_ptr {aka boost::shared_ptr<gtsam::noiseModel::Isotropic>}
In file included from /usr/include/boost/optional/optional.hpp:1597,
                 from /usr/include/boost/optional.hpp:15,
                 from /usr/local/include/gtsam/base/OptionalJacobian.h:25,
                 from /usr/local/include/gtsam/base/Matrix.h:25,
[... 127 lines cut ...]
                 from /usr/include/boost/lexical_cast.hpp:32,
                 from /usr/include/boost/date_time/string_parse_tree.hpp:13,
                 from /usr/include/boost/date_time/period_parser.hpp:14,
                 from /usr/include/boost/date_time/date_facet.hpp:23,
                 from /usr/include/boost/date_time/gregorian/gregorian_io.hpp:16,
                 from /usr/include/boost/date_time/gregorian/gregorian.hpp:31,
                 from /usr/include/boost/date_time/posix_time/time_formatters.hpp:12,
                 from /usr/include/boost/date_time/posix_time/posix_time.hpp:24,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:26,
                 from /home/doongli/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp:6:
[... cut ...]
      |                                           ^~~~~~~~~~~~~
make[2]: *** [direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/build.make:245: direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/preprocess/dynamic_point_cloud_integrator.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:990: direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j36 -l36" failed

koide3, 2023-09-25:
The error stems from the inconsistency of GTSAM versions. Please make sure that you are using the specified version `4.2a9` of GTSAM.

https://koide3.github.io/direct_visual_lidar_calibration/installation/

DoongLi, 2023-09-25:
I re-compiled GTSAM and Ceres，and it's work.

closed 2023-09-25 by DoongLi