koide3/direct_visual_lidar_calibration #54: Results are poor ! Why?

calibratorterminator55, 2023-09-22:
***Thank you for your wonderful research***

**Before opening an issue**
If the issue is about build errors:
- [x] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [ ] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [x] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [x] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [x] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**

- We do not obtain a good enough initial transform via SuperGlue. Why?

- We executed the calibrator of our bags but the NID result came out approximately 0.95. Then we assume our bags are not correctly package's requirements so we tried node with your bags. But the result was not perfect(proposed). Has anyone tried and obtained perfect calibration results it? 
- We have a robot so we cannot move to the vertical line. But we drive 10-15 second to generate dense point cloud. Could this be the source of the error?
or 
- LiDAR and Camera pose are not near as your setup. Is this not a good calibration result?

- Another problem is, that we are not an example bag's calibration ***ground truth***. We can not fully understand the result in your visualizer. We would like to review in Rviz.

- ***Lastly, Has anyone obtained a perfect result?***

**To Reproduce**
Steps to reproduce the behavior:
1. [ Go to '...'](https://koide3.github.io/direct_visual_lidar_calibration/)
2. We followed all the steps exactly in the documentation.

**Screenshots and sample data**

***The result of the experiment with your bag(2 bag)***

[image]

**Environment:**
 - OS: [e.g. Ubuntu 20.04]
 - ROS version: [e.g., ROS1 noetic

calibratorterminator55, 2023-09-25:
`How it failed?`
-  We have a 360 degree Ouster LiDAR. We are facing this problem:

[image]
The initial guess estimation in our case was **always** a failure. It is full of red outliers. Why this is happening? In the paper, you stated that Ouster Init. can also achieve good initial estimation.

`What does perfect mean? It is a quite vague word, and I cannot figure out what is happening on your side.`
 
- With "perfect calibration" we mean the results coming closer to your paper results. 
   Ouster Results: Avg. Trans error: 0.069, Rotation error: 0.724
Our results were far from these results so that's why we asked if we did something wrong. Our calibration is looking like this issue:
[image]
   [https://github.com/koide3/direct_visual_lidar_calibration/issues/48](url)  

`For velodyne-like LiDARs, a motion in the vertical direction is required. Please consider detaching the sensor from the robot and give a 6DoF motion for better calibration.`
- If we required to detach the lidar from the robot, what is the point of using targetless calibration toolbox? We can acquire a better calibration with checkers ex.. Is there any chance to overcome this problem?

-  `Possibly, you can publish the transformation T_lidar_camera via TF to visualize the result on rviz.`
-  Okay, thank you for your answer  :100:  We will try that soon as possible.

- We think there may be a problem with NID calculation and try with a lot of different NID registration parameters but the results were not that much affective. Sadly, we were not able to reproduce the suggested results with Ouster LiDAR.

Thank you for your attention and time for our questions. :pray:

koide3, 2023-10-26:
> How can we improve the results of Superglue using Ouster? Can we adapt a cropping pre-process so that it won't match the back of the robot?

In some situations, correspondence estimation can be ill-posed and becomes impossible even with SuperGlue. I think we can do only take data in environments with rich geometries so that the data contains enough features to establish correspondences.

> Are we required to detach the LiDAR from the robot in Ouster LiDAR calibration? Is there any chance to overcome this problem?

I think there is no other ways to do calibration without calibration targets. If you cannot detach the sensor from the robot, I recommend using target-based algorithms.