koide3/direct_visual_lidar_calibration #55: How to set parameters

Eric5709, 2023-09-27:
**Thanks for your great research!**

I have two questions.

1. I've heard that even small change can lead to significant differences in sensor calibration.
When I adjust the parameter `robust_kernel_width` in function  _"estimate_rotation_ransac"_ at _"estimate_pose.cpp"_, result changes significantly. So it seems necessary to determine the correct parameters for different dataset.
Could you please provide guidance on how to set appropriate parameters like `robust_kernel_width` or `error_thresh_sq`?

2. I've noticed a step that identifies `inliers` in the "estimate_rotation_ransac" function, but it appears that this information is never utilized. I'm curious why the code estimates the pose using all correspondences rather than just the `inliers` in the _"estimate_pose_lsq"_ function.

Thanks!

koide3, 2023-10-26:
1. `robust_kernel_width` changes the sensitivity to outliers in 2D-3D correspondences. If your data contains many wrong correspondences decrease this parameter to filter out them.
2. Because we use robust kernels in the LSQ optimization, we can expect that outliers are automatically removed. We thus simply gave it all correspondences.

Eric5709, 2023-12-15:
Thank you for the kind response!

closed 2023-12-15 by Eric5709