koide3/direct_visual_lidar_calibration #57: Double free or corruption (out) Aborted (core dumped)

Tony-TF, 2023-10-09:
Hello author, thank you very much for sharing your current work, but the following problems occurred when reproducing your work. When using the Oster data you provided, I manually filled in the camera parameters according to the question prompts, but there were related problems with the function. The Camera parameters are given according to    calib.json example。

The Environment : UBuntu 18.04  Ros:Melodic

Input :
rosrun direct_visual_lidar_calibration preprocess --data_path /home/zlzk/catkin_ws/ouster  --dst_path /home/zlzk/catkin_ws/ouster_preprocessed --image_topic /image  --points_topic /points  --camera_model plumb_bob --camera_intrinsics 1452.71,1455.87,1265.25,1045.81  --camera_distortion_coeffs  -0.0403,0.087,0.002,0.005  -adv

Result:
[image]

The Environment : UBuntu 20.04  Ros:Noetic

Input :
rosrun direct_visual_lidar_calibration preprocess --data_path /home/zlzk/catkin_ws/ouster  --dst_path /home/zlzk/catkin_ws/ouster_preprocessed --image_topic /image  --points_topic /points  --camera_model plumb_bob --camera_intrinsics 1452.71,1455.87,1265.25,1045.81  --camera_distortion_coeffs  -0.0403,0.087,0.002,0.005  -adv

Result:
[image]

I have also seen related issues. Using the dataset they provided and switching to the vlp16 branch（https://github.com/koide3/direct_visual_lidar_calibration/issues/13）, the same problem still exists.

[image]

koide3, 2023-10-26:
Does the error occur when using the docker images?
https://koide3.github.io/direct_visual_lidar_calibration/docker/

Tony-TF, 2023-10-26:
There are no problems when using Docker, but I want to deploy it locally to facilitate debugging.

koide3, 2023-10-26:
If the error does not occur with docker images, it is likely caused by inconsistent build options for dependent libraries (e.g., self-built PCL with `march=native` option). Can you see if the error persists when using libraries installed via apt?

Tony-TF, 2023-10-30:
Thank you for your reply. I have solved the problem by switching to Ubuntu20.04 system.

closed 2023-10-30 by Tony-TF

zhh2005757, 2024-11-05:
Maybe my solution can help #123