koide3/direct_visual_lidar_calibration #59: Nvidia Jetson support

gyillikci, 2023-10-25:
Hi,

Thanks for the tool. When pool the docker for Humble, I get the error of architectural mismatch. It would be very beneficial to have a arm64 support. Maybe a guideline for Jetson devices would be an alternative solution.

koide3, 2023-10-26:
Thanks for your request. I tried to add ARM64 support in the past, but I found that it required a very tedious CI setup. While I would like to avoid ARM64 support, I believe you can build the provided docker images on Jetson devices without errors.

https://github.com/koide3/direct_visual_lidar_calibration/tree/main/docker

gyillikci, 2023-11-01:
Hello Kenji,

I do the comping on Jetson Jetpack 5.1.2. with ROS2 Foxy. 
I got a problem with rosba2_cpp as below wile running colcon build. 
[image]
Do you have any suggestions if possible.

gyillikci, 2023-11-01:
By the way when I run the build I get architecture error

[image]

koide3, 2023-11-13:
Ah, the base image is built for only AMD64. Please build an image for ARM64 using Dockerfiles in the following repository.

https://github.com/koide3/gtsam_docker

annxjz, 2023-12-14:
> Hello Kenji,
>
> I do the comping on Jetson Jetpack 5.1.2. with ROS2 Foxy. I got a problem with rosba2_cpp as below wile running colcon build. [...]

I encountered the same problem when I compiled it from source (ROS2 FOXY), did you solve the problem? Do you have any suggestions ?

annxjz, 2023-12-14:
> Hello Kenji,
>
> I do the comping on Jetson Jetpack 5.1.2. with ROS2 Foxy. I got a problem with rosba2_cpp as below wile running colcon build. [...]

I encountered the same problem when I compiled it from source (ROS2 FOXY, x64 architecture, 20.04), did you solve the problem? Do you have any suggestions ?

ginop-1, 2024-04-24:
> I encountered the same problem when I compiled it from source (ROS2 FOXY, x64 architecture, 20.04), did you solve the problem? Do you have any suggestions ?

Same here, having some problem while building the package on foxy, ubuntu 20.04. I installed every dependency but it seems like fmt is not being installed properly (`fatal error: fmt/format.h: No such file or directory`)

poornimajd, 2024-07-03:
@enricocovili I am getting the same error, were you able to resolve it?