koide3/direct_visual_lidar_calibration #65: An error occured when I run hte preprocess program

Lan-hhxx, 2023-12-15:
Hello, I tried to reappear your algorithm, however some error occured when I run the preprocess program using your bags.
the command and the result are as fellows:
rosrun direct_visual_lidar_calibration preprocess ../direct_data/livox_ros1/ ../preprocessed/ -av
data_path: ../direct_data/livox_ros1/
dst_path : ../preprocessed/
input_bags:
- ../direct_data/livox_ros1/rosbag2_2023_03_09-13_42_46.bag
- ../direct_data/livox_ros1/rosbag2_2023_03_09-13_44_54.bag
- ../direct_data/livox_ros1/rosbag2_2023_03_09-13_44_10.bag
- ../direct_data/livox_ros1/rosbag2_2023_03_09-13_46_10.bag
- ../direct_data/livox_ros1/rosbag2_2023_03_09-13_46_54.bag
topics in ../direct_data/livox_ros1/rosbag2_2023_03_09-13_42_46.bag:
- /livox/points : sensor_msgs/PointCloud2
- /livox/imu : sensor_msgs/Imu
- /image : sensor_msgs/Image
- /camera_info : sensor_msgs/CameraInfo
selected topics:
- camera_info: /camera_info
- image      : /image
- points     : /livox/points
intensity_channel: intensity
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: /image
Aborted (core dumped)

I don't know how to solve it, so please help me to solve this issue, besides,  I wonder  confirm the version of Dependencies such as openCV. 
 thank you!

Seekerzero, 2024-02-12:
I think this is because you use a different version of opencv. The verified version of openCV is 4.2.0, please check if you get the build warning when you run catkin_make for the rospackage. similar to this:
```shell
warning: libopencv_core.so.408, needed by /your/path/to/the/catkin_ws/devel/lib/libdirect_visual_lidar_calibration.so, may conflict with libopencv_core.so.4.2
```
In this warning it says that the openCV 4.8.0 has some incompatible files to the openCV 4.2.0.

BreCaspian, 2025-02-20:
I am facing the same problem and I can not slove it ,can you help me?

```
yao@RM:~/catkin_ws$ rosrun direct_visual_lidar_calibration preprocess -av --camera_model plumb_bob --camera_intrinsic 1452.0,1453.1,0.2897,0.2058 --camera_distortion_coeffs 0.1392,-1.6770,0.0,0.0,0.0 /media/yao/RM/rosbag_test /media/yao/RM/rosbag_preprocessed
data_path: /media/yao/RM/rosbag_test
dst_path : /media/yao/RM/rosbag_preprocessed
input_bags:
- /media/yao/RM/rosbag_test/lidar_camera_data_test1.bag
topics in /media/yao/RM/rosbag_test/lidar_camera_data_test1.bag:
- /cloudpoints : sensor_msgs/PointCloud2
- /camera/camera_info : sensor_msgs/CameraInfo
- /camera/image : sensor_msgs/Image
selected topics:
- camera_info: /camera/camera_info
- image      : /camera/image
- points     : /cloudpoints
intensity_channel: intensity
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error opening file: /camera/image
Aborted (core dumped)

```