koide3/direct_visual_lidar_calibration #66: Omnidirectional camera calibration

Eric5709, 2023-12-26:
Hi, thanks for your work and open source code. It is really helpful. First of all, I'm not familiar with omni-directional camera. I utilized the same camera model from the paper, Kodak PixPro 4KVR360 and some questions arised when using it:

- [x] I've done calibration using other pinhole camera model(d435i).

1. The camera uses two different cameras to cover 360 degree, which means that we need two intrinsic parameters for each camera for the process(ex: reprojection, ...) but in your code, you used only one intrinsic parameter in _"/include/camera/omnidir.hpp."_ My question  is that "if you have two cameras, shouldn't you use two parameters?".
2. The next thing I'm curious about is that the method you obtain the intrinsic parameter or relative pose of two cameras. There is no tool for Kodak PixPro 4KVR360 to obtain parameters as far as I know.

Thank you for your time.

koide3, 2023-12-27:
1. This package only does LiDAR-camera (point-cloud-to-image) calibration. So, if you want the transformation between two cameras, you first need to estimate the transformation between a point cloud frame and both the image frames and then combine these results to find the transformation between the omnicam and the realsense.
2. In case of the omnidirectional camera, you don't need to specify intrinsic parameters since the intrinsic params consist of only image width and height.

Eric5709, 2023-12-27:
Thank you for your fast response!

I think I may have misrepresented the question. To explain specifically, I'm facing the problem when turning on the omnidirecitonal camera in ROS. I couldn't find any github repository or other tool for the camera model(Kodak PixPro 4KVR360) so I've modified the repository which is targeting on different model(Kodak sp360) which utilizes only one camera. The output image seems to be misaligned at the boundary of two cameras(the front and the doom) so I thought that there might to some kind of preprocess before the LiDAR-camera extrinsic calibration.(there is software(PIXPRO 360 VR SUITE) that is provided by Kodak for calibrating the front camera and doom camera in Mac/Windows)
To obtain rosbag, you must have turned on the camera through ROS I was wondering how you did it.

And other question was that when projecting 3D point into 2D image point, only one intrinsic matrix(fx, fy, cx, cy, xi) and one distortion vector(k1, k2, p1, p2) are used in 
_https://github.com/koide3/direct_visual_lidar_calibration/blob/main/include/camera/omnidir.hpp_

```
    const auto& fx = intrinsic[0];
    const auto& fy = intrinsic[1];
    const auto& cx = intrinsic[2];
    const auto& cy = intrinsic[3];
    const auto& xi = intrinsic[4];

    const auto& k1 = distortion[0];
    const auto& k2 = distortion[1];
    const auto& p1 = distortion[2];
    const auto& p2 = distortion[3];
```
Since two cameras cover different areas, I wonder if I should use the intrinsic parameter of the camera corresponding to the area when projecting the point.

If there is any chance, can I get the rosbag of Kodak pixPro 4KVR360 and Ouster that is posted on the youtube?

> The repository I've modified to turn on the camera in ROS:
> [https://github.com/uts-magic-lab/kodak_pixpro_sp360_ros](url)
> 

> The following link is the Kodak software site:
> [https://kodakpixpro.com/support/downloads/ 
> ](url)

Eric5709, 2024-01-03:
I got it! I was just trying wrong camera model.
Since Kodak pixpro 4KVR360's the output image is equirectangular, I need to try 'equirectangular' camera model not 'omnidirectional' camera model. The 'omnidirectional' camera model is for the traditional omnidirectional camera like catadioptric camera. Thanks a lot!

closed 2024-01-03 by Eric5709