koide3/direct_visual_lidar_calibration #69: distortion coefficients k4 k5 k6 support

lddnb, 2024-01-19:
Hi, thanks for this great project.
I want to use distortion coefficients of [k1, k2, p1, p2, k3, k4, k5, k6], but it only use [k1, k2, p1, p2, k3] in program.

I tried to change 
`const auto r_coeff = 1.0 + k1 * r2 + k2 * r4 + k3 * r6;`
to 
`const auto r_coeff = (1.0 + k1 * r2 + k2 * r4 + k3 * r6)/(1.0 + k4 * r2 + k5 * r4 + k6 * r6);`
in `include/camera/pinhole.hpp`

But the projection result is not correct.

When I use the 8 elements distortion coefficients with OpenCV, I can obtain correct undistorted images, so I think it is nothing wrong with these coefficients.

Please you give me some suggestions to  solve it.
Thank you!

FLYNGU, 2024-12-02:
In my opinion, I have a idea to skip this part.  you can record data with undistoted image, and you can calculated a new intrinsic matrix by 'cv::getOptimalNewCameraMatrix' , so that the coefficients is [0, 0, 0, 0 ,0].

iceriver97, 2024-12-02:
> In my opinion, I have a idea to skip this part. you can record data with undistoted image, and you can calculated a new intrinsic matrix by 'cv::getOptimalNewCameraMatrix' , so that the coefficients is [0, 0, 0, 0 ,0].

感谢回复，我稍后会试一下。