koide3/direct_visual_lidar_calibration #71: calibration_resualt is poor than initial_guess

Chatoyant19, 2024-02-26:
Hi, I used data from your [faro.tar.gz](https://zenodo.org/record/7780490/files/faro.tar.gz?download=1).But my initial guess result is better than refine calibration reult.
initial guess projection:
[image]
refine calib projection:
[image]

When run calibrate:
[image]
I hope your help! Thanks!

liucsg, 2024-06-21:
I also encountered the same problem, did you solve it?

Chatoyant19, 2024-07-10:
> I also encountered the same problem, did you solve it?

The problem still exists when I use this dataset. However, when I use other datasets and provide a relatively good initial extrinsics, some calibrations are successful, but some still fail. I suspect that lighting or some other factors might have a significant impact on the results.