koide3/direct_visual_lidar_calibration #79: Ros2 Support for Foxy Distro

jeychandar, 2024-04-16:
I follow the installation and tried to build for ros2 foxy but it throws error. Can you help me out to resolve this issue. Thanks in advance
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note:   candidate expects 1 argument, 0 provided
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp: In instantiation of ‘std::shared_ptr<_Tp> vlcal::PreprocessROS2::get_first_message(const string&, const string&) const [with T = sensor_msgs::msg::Image_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>]’:
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:108:98:   required from here
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:54:25: error: no matching function for call to ‘rosbag2_cpp::Reader::Reader()’
   54 |     rosbag2_cpp::Reader reader;
      |                         ^~~~~~
In file included from /home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:4:
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note: candidate: ‘rosbag2_cpp::Reader::Reader(std::unique_ptr<rosbag2_cpp::reader_interfaces::BaseReaderInterface>)’
   53 |   explicit Reader(std::unique_ptr<reader_interfaces::BaseReaderInterface> reader_impl);
      |            ^~~~~~
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note:   candidate expects 1 argument, 0 provided
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp: In instantiation of ‘std::shared_ptr<_Tp> vlcal::PreprocessROS2::get_first_message(const string&, const string&) const [with T = sensor_msgs::msg::CompressedImage_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>]’:
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:112:106:   required from here
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:54:25: error: no matching function for call to ‘rosbag2_cpp::Reader::Reader()’
   54 |     rosbag2_cpp::Reader reader;
      |                         ^~~~~~
In file included from /home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:4:
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note: candidate: ‘rosbag2_cpp::Reader::Reader(std::unique_ptr<rosbag2_cpp::reader_interfaces::BaseReaderInterface>)’
   53 |   explicit Reader(std::unique_ptr<reader_interfaces::BaseReaderInterface> reader_impl);
      |            ^~~~~~
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note:   candidate expects 1 argument, 0 provided
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp: In instantiation of ‘std::shared_ptr<_Tp> vlcal::PreprocessROS2::get_first_message(const string&, const string&) const [with T = sensor_msgs::msg::CameraInfo_<std::allocator<void> >; std::string = std::__cxx11::basic_string<char>]’:
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:117:113:   required from here
/home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:54:25: error: no matching function for call to ‘rosbag2_cpp::Reader::Reader()’
   54 |     rosbag2_cpp::Reader reader;
      |                         ^~~~~~
In file included from /home/praveen/Lidar_camera_fusion/src/direct_visual_lidar_calibration/src/preprocess_ros2.cpp:4:
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note: candidate: ‘rosbag2_cpp::Reader::Reader(std::unique_ptr<rosbag2_cpp::reader_interfaces::BaseReaderInterface>)’
   53 |   explicit Reader(std::unique_ptr<reader_interfaces::BaseReaderInterface> reader_impl);
      |            ^~~~~~
/opt/ros/foxy/include/rosbag2_cpp/reader.hpp:53:12: note:   candidate expects 1 argument, 0 provided
make[2]: *** [CMakeFiles/preprocess.dir/build.make:76: CMakeFiles/preprocess.dir/src/preprocess_ros2.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:256: CMakeFiles/preprocess.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< direct_visual_lidar_calibration [7.06s, exited with code 2]

Summary: 2 packages finished [7.43s]
  1 package failed: direct_visual_lidar_calibration
  1 package had stderr output: direct_visual_lidar_calibration

**Environment:**
 - OS: [ Ubuntu 20.04]
 - ROS version: [e.g., ROS2 foxy]

[image]

koide3, 2024-04-16:
Sorry, we have no plan to support ROS2 foxy. But you can still use Humble docker image on Ubuntu 20.04.

jeychandar, 2024-04-17:
Thanks for your prompt reply :)

closed 2024-04-17 by jeychandar