koide3/direct_visual_lidar_calibration #80: How to publish coloured point-cloud in real-time?

jeychandar, 2024-04-17:
1. Post lidar-camera calibration, I need to publish processed coloured pointcloud data and lidar points projected on image as ros sensor messages in real time. I am unable to find where it is publishing in code. Is it publishing? If it is, can you please help me out where it is publishing. 

2. Given a (x,y) coordinate on image, how can I extract XYZ information of corresponding points in the point cloud data? (i.e., p_lidar = T_lidar_camera * p_camera). Should I be using this equation for this? may i know where it is implemented in code?

Thanks in advance.

koide3, 2024-04-25:
1. This package only estimates the LiDAR-camera transformation and do not colorize nor publish point clouds.
2. If you are looking for a way to project LiDAR points on the image, the projection implementation can be found at https://github.com/koide3/direct_visual_lidar_calibration/blob/main/include/camera/pinhole.hpp

jeychandar, 2024-04-25:
@koide3 Thanks for your reply. However there is some offset while projecting LiDAR points on the image, but when open in viewer it seems correct and the offset logic is handled somewhere and I could not able to see where it is. Could you please tag that reference line.

closed 2024-04-26 by jeychandar