koide3/direct_visual_lidar_calibration #81: Hello，I'm having problems running docker's preprocessing

ZhangYBZ, 2024-04-19:
When I was running docker preprocessing, I had a problem uploading a local packet. 
Under this directory is the Calibration example package, which includes a.yaml file, a.db3 file, and a.bag file

docker run   
--rm   
--net host   
--gpus all   
-e DISPLAY=$DISPLAY  
-v $HOME/.Xauthority:/root/.Xauthority   
-v 1rosbag2:/tmp/input_bags   
-v out:/tmp/preprocessed   
koide3/direct_visual_lidar_calibration:humble   ros2 run direct_visual_lidar_calibration preprocess -a -v /tmp/input_bags /tmp/preprocessed

tmp/input_bags /tmp/preprocessed
data_path: /tmp/input_bags
dst_path : /tmp/preprocessed
input_bags:
error: no input bags!!

koide3, 2024-04-24:
Maybe you need to specify input paths as absolute paths.

ZhangYBZ, 2024-04-24:
**When absolute paths are used, a new problem arises**
-$ docker run   --rm   --net host   --gpus all   -e DISPLAY=$DISPLAY   -v $HOME/.Xauthority:/root/.Xauthority   -v /home/zzy/documents/livox/1rosbag2:/tmp/input_bags   -v /home/zzy/documents/livox/out:/tmp/preprocessed   koide3/direct_visual_lidar_calibration:humble   ros2 run direct_visual_lidar_calibration preprocess -a -v /tmp/input_bags /tmp/preprocessed
data_path: /tmp/input_bags
dst_path : /tmp/preprocessed
[ERROR] [1713959106.825795775] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI
[INFO] [1713959106.827064691] [rosbag2_storage]: Opened database '/tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3' for READ_ONLY.

closing.

closing.

closing.

closing.
input_bags:
- /tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3
[INFO] [1713959106.830188673] [rosbag2_storage]: Opened database '/tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3' for READ_ONLY.

closing.

closing.
topics in /tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3:
- /camera_info : sensor_msgs/msg/CameraInfo
- /image : sensor_msgs/msg/Image
- /livox/imu : sensor_msgs/msg/Imu
- /livox/lidar : livox_interfaces/msg/CustomMsg
- /livox/points : sensor_msgs/msg/PointCloud2
selected topics:
- camera_info: /camera_info
- image      : /image
- points     : /livox/points
[INFO] [1713959106.833216940] [rosbag2_storage]: Opened database '/tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3' for READ_ONLY.

closing.

closing.
intensity_channel: intensity
[INFO] [1713959106.843135915] [rosbag2_storage]: Opened database '/tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3' for READ_ONLY.

closing.

closing.
try to get the camera model automatically
[INFO] [1713959106.869106008] [rosbag2_storage]: Opened database '/tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3' for READ_ONLY.

closing.

closing.
camera_model: plumb_bob
image_size  : 2448 2048
intrinsics  : 1452.71 1455.88 1265.26 1045.82
dist_coeffs : -0.0420356  0.0873171 0.00238638  0.0056297 -0.0425115
processing images and points (num_threads_per_bag=20)
start processing /tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3
[INFO] [1713959106.871861839] [rosbag2_storage]: Opened database '/tmp/input_bags/rosbag2_2023_03_09-13_42_46_0.db3' for READ_ONLY.
Authorization required, but no authorization protocol specified
glfw error 65544: X11: Failed to open display :1
failed to initialize GLFW

closing.

closing.
[ros2run]: Segmentation fault

koide3, 2024-04-25:
It seems ROS2 bag format issue is occurring. What is the format of your rosbag?

iceriver97, 2024-12-03:
> It seems ROS2 bag format issue is occurring. What is the format of your rosbag?
I have the same problem;
```
wq@wq-PC:~/direct_lidar_camera$ ros2 run direct_visual_lidar_calibration preprocess ./1202-dyn/ ./1202-dyn-preprocessed-manual/ --image_topic camera/image_raw/compressed --points_topic /rslidar_points -dv
data_path: ./1202-dyn/
dst_path : ./1202-dyn-preprocessed-manual/

closing.

closing.
[ERROR] [1733204874.165103002] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI

closing.

closing.
[INFO] [1733204874.166085087] [rosbag2_storage]: Opened database './1202-dyn/1202-dyn.db3' for READ_ONLY.
input_bags:
- ./1202-dyn/1202-dyn.db3
warning: failed to get camera_info topic!!
selected topics:
- camera_info: 
- image      : camera/image_raw/compressed
- points     : /rslidar_points

closing.

closing.
[INFO] [1733204874.168155541] [rosbag2_storage]: Opened database './1202-dyn/1202-dyn.db3' for READ_ONLY.
intensity_channel: intensity

closing.

closing.
[INFO] [1733204874.191100226] [rosbag2_storage]: Opened database './1202-dyn/1202-dyn.db3' for READ_ONLY.
try to get the camera model automatically
camera_model: plumb_bob
image_size  : 3840 2160
intrinsics  : 1147.33 1675.26 1916.63 1079.76
dist_coeffs : 0 0 0 0 0
processing images and points (num_threads_per_bag=20)
start processing ./1202-dyn/1202-dyn.db3

closing.

closing.
[INFO] [1733204874.236322440] [rosbag2_storage]: Opened database './1202-dyn/1202-dyn.db3' for READ_ONLY.

closing.

closing.
[INFO] [1733204874.262959739] [rosbag2_storage]: Opened database './1202-dyn/1202-dyn.db3' for READ_ONLY.
warning: per-point timestamps are not given!!
       : use pseudo per-point timestamps based on the order of points
[ros2run]: Segmentation fault
```
baginfo:
```
Files:             1202-dyn.db3
Bag size:          982.0 MiB
Storage id:        sqlite3
Duration:          74.884508891s
Start:             Dec  2 2024 10:17:40.535491185 (1733105860.535491185)
End:               Dec  2 2024 10:18:55.420000076 (1733105935.420000076)
Messages:          409
Topic information: Topic: /rslidar_points | Type: sensor_msgs/msg/PointCloud2 | Count: 408 | Serialization Format: cdr
                   Topic: camera/image_raw/compressed | Type: sensor_msgs/msg/CompressedImage | Count: 1 | Serialization Format: cdr

```

themightyoarfish, 2024-12-12:
Same problem, whats the fix here?