koide3/direct_visual_lidar_calibration #82: set_model_matrix error during build

ginop-1, 2024-04-29:
**Before opening an issue**
If the issue is about build errors:
- [x] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [x] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [ ] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [ ] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [ ] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
I installed every dependency (superglue excluded) and I get this error while building the package in my ROS2 workspace:
(docker is not an option in my case for a variety of reasons)
```
error: call of overloaded ‘set_model_matrix(Eigen::DenseBase<Eigen::Matrix<double, 4, 4> >::CastXpr<float>::Type)’ is ambiguous
  191 |           T_lidar_camera_gizmo.set_model_matrix(T_camera_lidar->inverse().matrix().cast<float>());
      |           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/enri/mmr-drive/src/1_perception/direct_visual_lidar_calibration/src/initial_guess_manual.cpp:16:
/usr/local/include/iridescence/guik/model_control.hpp:29:8: note: candidate: ‘void guik::ModelControl::set_model_matrix(const Matrix4f&)’
   29 |   void set_model_matrix(const Eigen::Matrix4f& matrix) { pose = Eigen::Affine3f(matrix); }
      |        ^~~~~~~~~~~~~~~~
/usr/local/include/iridescence/guik/model_control.hpp:30:8: note: candidate: ‘void guik::ModelControl::set_model_matrix(const Matrix4d&)’
   30 |   void set_model_matrix(const Eigen::Matrix4d& matrix) { pose = Eigen::Affine3f(matrix.cast<float>()); }

```
**To Reproduce**
Steps to reproduce the behavior:
Install all dependencies
build the package in my ROS2 workspace 

**Expected behavior**
Successful build 

**Screenshots and sample data**

**Environment:**
 - OS: Ubuntu 22.04.4 LTS
 - ROS version: ROS2 Humble

ginop-1, 2024-04-29:
SOLVED: just changed the order of the call to: `T_camera_lidar->inverse().cast<float>().matrix()` to match what the compiler expects, as previously the matrix was not loaded yet

closed 2024-04-29 by ginop-1

koide3, 2024-04-30:
Thanks for reporting the issue. It might be caused by an update in the viewer library. I'll check and fix it.

ginop-1, 2024-04-30:
> Thanks for reporting the issue. It might be caused by an update in the viewer library. I'll check and fix it.

No problem. Thank you for your work.

cyhunblr, 2024-05-06:
> SOLVED: just changed the order of the call to: `T_camera_lidar->inverse().cast<float>().matrix()` to match what the compiler expects, as previously the matrix was not loaded yet

This problem also exists for ROS1 and fixed like this. Thanks..

imanity04, 2025-03-23:
Hello, may I ask if you have encountered the following problems during pre-processing?"
terminate called after throwing an instance of 'cv::Exception'
What () : OpenCV (4.10.0)/home/khadas/Downloads/OpenCV - 4.10.0 / modules/core/SRC/matrix. The CPP: 246: error: (-215:Assertion failed) s >= 0 in function 'setSize' ”
How to solve it, thank you!

imanity04, 2025-03-23:
Hello, may I ask if you have encountered the following problems during pre-processing?
" terminate called after throwing an instance of 'cv::Exception'
What () : OpenCV (4.10.0)/home/khadas/Downloads/OpenCV - 4.10.0 / modules/core/SRC/matrix. The CPP: 246: error: (-215:Assertion failed) s >= 0 in function 'setSize' ”
How to solve it, thank you!