koide3/direct_visual_lidar_calibration #84: Symbol lookup error when running preprocess node

Prabuddhi-05, 2024-05-23:
After a successful colcon build of all the installed packages, I encountered the following error when trying to run the preprocess node from the direct_visual_lidar_calibration package: This error seems to be related to a missing symbol in the libdirect_visual_lidar_calibration.so library, which is causing the preprocess node to fail.

Environment:
    ROS 2 Distribution: Humble
    Operating System: Ubuntu 22.04

Please help me in resolving this issue. 

[image]

Prabuddhi-05, 2024-05-28:
> I saw this error when I re-built GTSAM with different build options. In that case, the error resolved by re-building the package after removing `build` and `install` directories.

Thank you for the response. I tried doing it but I got the same error.

Seekerzero, 2024-06-13:
sometimes this happened when you didn't install the package into the system bin file after your edit and build but not intall it. Please make sure your all dependencies (like iridescence) is correctly build and install.