koide3/direct_visual_lidar_calibration #85: Failed to open blickfield_preprocessed_latest/xxx.bag_matches.json

jeychandar, 2024-05-29:
@koide3 Hi Sir, I have tried to create calib. json and other parameters like intensity, bag.ply using preprocess method and found out in the folder that it does not get intensity parameter from the bag. I could not find what is the issue. Could you help me out. Thanks in advance

[image]
[image]

jeychandar, 2024-06-13:
@koide3 Hi Sir, I looked through your code and unable to resolve this issue yet. Could you help me out where to check regarding this issue.

Seekerzero, 2024-06-13:
Hi jey, could you send your commandline for running the preprocess step to here. Also, an example of your lidar PointCloud2 fileds format would be helpful. Thanks.

jeychandar, 2024-06-13:
@Seekerzero Thank you for your reply Sir. I ran this command **rosrun direct_visual_lidar_calibration preprocess blickfield blickfield_preprocessed_check -av** and also tried **-i intensity** additionally with this command. I attached pointcloud2 format and publishing option below.

[image]

[image]

Seekerzero, 2024-06-13:
Did you get any error like this: "error: failed to determine point intensity channel automatically". If not, the intensity value for each points should be correctly saved into the ply files.

jeychandar, 2024-06-13:
I did not get any errors like what you have mentioned above Sir.

[image]

Seekerzero, 2024-06-13:
So everything should be correct. Like I said in the previous post, if you see the Turbo color from the viewer, the intensity value for each point is correctly loaded and saved. Please also check [this](https://github.com/koide3/direct_visual_lidar_calibration/blob/8a03c738573b2ec0148d8ee39709ac3948ed7608/src/vlcal/preprocess/preprocess.cpp#L248C1-L254C78) .

jeychandar, 2024-06-13:
what could be the reason that these images are empty? The preprocess step is failing to return .bag_matches.json. Does it have anything to do with lidar point cloud frame convention difference?

[image](https://github.com/koide3/direct_visual_lidar_calibration/assets/48254471/9ced7f66-4640-4dc1-a893-18b65180e22f)
[image]

Seekerzero, 2024-06-13:
The preprocess step should only return one json file, which is calib.json. 

Regarding to the intensity image, I would recommend first to check if the values are returned correctly [here](https://github.com/koide3/direct_visual_lidar_calibration/blob/8a03c738573b2ec0148d8ee39709ac3948ed7608/src/vlcal/preprocess/preprocess.cpp#L205C1-L205C136). You should see something like [0, ... , 0 , index] since you have a empty image. If there are all zeros for the intensities like I guess, you might look into the lidar_proj, otherwise, it could be something related to opencv.

jeychandar, 2024-06-25:
@Seekerzero As you mentioned earlier. I cross verified with some other tool  called opencalib/ calib_anything to check if the rosbag has intensity recorded. I am able to extract lidar intensity. since this repo direct_visual_calibration tool creates .ply file with intensity, I am able to analyse that lidar_proj have no issues. As you mentioned it could related to opencv I am not sure where could I able to verify that. Could you help me out to resolve this issue?. For your reference I am using blickfeld driver **https://git.tu-berlin.de/ecschuetz/ros_assembly/-/blob/main/src/ros_blickfeld_driver_src-v1.4.3/modules/ros_blickfeld_driver_core/src/blickfeld_driver_point_cloud_parser.cpp?ref_type=heads**

Seekerzero, 2024-06-25:
You should set a break point after the line of code that I linked in the previous post (or you basically just save the intensity values to a files) to see if the values are all zeros here. This is the first thing you should check. 
I believe it is not related to the lidar driver since the ply file are successfully generated and verified.

jeychandar, 2024-06-25:
@Seekerzero I have saved the intensity values into text files and it returned zeros's.

[image]

jeychandar, 2024-06-25:
@Seekerzero I have attached two images below one for livox(From the rosbag given in this repo) and another for blickfeld sensor. You can notice the frame convention difference. Do you think it impacts the LiDAR projection?. 

**Livox Sensor:**

[image]

**Blickfeld sensor:**

[image]

Seekerzero, 2024-06-25:
yup, I believe so. Since the projection range is calculated based on [here](https://github.com/koide3/direct_visual_lidar_calibration/blob/8a03c738573b2ec0148d8ee39709ac3948ed7608/src/vlcal/preprocess/preprocess.cpp#L188), and given the fov that calculated from your sensor 72 degree, these points might get all filtered.

jeychandar, 2024-06-25:
@Seekerzero Yes true it is because of the frame convention. I am able to get lidar intensity image and also intensity values in the somewhere in the middle have values and other values are zero. I tried with this command **rosrun direct_visual_lidar_calibration initial_guess_auto blickfield_preprocessed_check** 

It throws error like this still
loading blickfield_preprocessed_check/2024-06-08-12-48-43.bag.(png|ply)
error: failed to open blickfield_preprocessed_check/2024-06-08-12-48-43.bag_matches.json
Aborted (core dumped)

[image]

Seekerzero, 2024-06-25:
after run preprocess, you should run
`rosrun direct_visual_lidar_calibration find_matches_superglue.py <your preprocessed folder>` before you run the initial_guess_auto.

jeychandar, 2024-06-25:
@Seekerzero Yes correct working able to get extrinsic matrix thank you for your guidance. I have another question don't you feel there is some offset in superglue generated png? Let me try this extrinsic value first.

[image]

Seekerzero, 2024-06-25:
superglue might not work if the lidar image does not get enough similarities to the image capture (not enough density). You might want to look into the slam approach to accumulate the lidar image. Please look into the document example and check that.

koide3, 2024-06-26:
@jeychandar 
As @Seekerzero said, for SuperGlue-based alignment, the point cloud needs enough density to generate a LiDAR image with good similarity with the camera image. Try the dynamic points accumulator to densify the point cloud.

@Seekerzero 
Thanks always for your help!

jeychandar, 2024-06-26:
@Seekerzero @koide3  As you were mentioned earlier I referred the documentation and collected the sample data Up and Down for longer rosbag record since blickfeld LiDAR is repetitive. I tried the dynamic points accumulator to densify the point cloud by using this command **rosrun direct_visual_lidar_calibration preprocess blickfield blickfield_preprocessed_check -adv** . It really helped to extract the intensity Image and extrinsic matrix. Finally I able to test Lidar_camera_fusion with resulted extrinsic matrix and it is accurate. Once again Thank you for your support! @koide3 and @Seekerzero

closed 2024-06-26 by jeychandar