koide3/direct_visual_lidar_calibration #86: Disable translation estimation when calibration

ClementLeBihan, 2024-06-21:
**Is your feature request related to a problem? Please describe.**
The final camera_lidar translation is wrong, but the projection is pretty correct. As I already know (+- 5cm) the position of the camera, I would love to do a first step of optimization computing only rotations. And when we reach the best estimation, start from this to refine the translations+rotations. I hope it will decrease the translation error. 

**Describe the solution you'd like**
I would love to know how to force the optimizer to don't optimize translations. Only rotations.

koide3, 2024-06-22:
I should be possible by adding a factor that constraints the translation components around the initial position at the following optimization part. 

https://github.com/koide3/direct_visual_lidar_calibration/blob/8a03c738573b2ec0148d8ee39709ac3948ed7608/src/vlcal/calib/visual_camera_calibration.cpp#L216

sriram487, 2024-12-19:
> **Is your feature request related to a problem? Please describe.** The final camera_lidar translation is wrong, but the projection is pretty correct. As I already know (+- 5cm) the position of the [...]

@ClementLeBihan Are you able to fix the wrong translation. Even I am able to achieve the correct projections but the final camera_lidar translation is not that accurate.