koide3/direct_visual_lidar_calibration #88: Error glfw

Davabbb, 2024-06-25:
glfw error 65544: X11: Failed to open display :2
failed to initialize GLFW

i try xhost local:root and export DISPLAY=:...

Davabbb, 2024-06-25:
@Seekerzero

Seekerzero, 2024-06-25:
Are you running this in a virtual machine? Currently glfw needs a physical graphic card to initialize.

Davabbb, 2024-06-25:
no

koide3, 2024-06-26:
Can you run `glxinfo`? If not there should be X-related issues that may be unrelated to the calibration program.

Davabbb, 2024-06-26:
[image]

Davabbb, 2024-06-26:
[image]

Davabbb, 2024-06-26:
![image](https://github.com/koide3/direct_visual_lidar_calibration/assets/57668741/80dc39a9-1c7d-4999-b59f-34de96bcff42
)

closed 2024-07-03 by Davabbb