koide3/direct_visual_lidar_calibration #91: Poor results with example dataset

khuechuong, 2024-07-02:
I tried the code with the ros1 example dataset. And I got pretty bad results:
```
rosrun direct_visual_lidar_calibration viewer ouster_preprocessed
Automatic initial guess result found
--- T_lidar_camera ---
 0.492723  0.166985 -0.854014   0.39141
-0.326018  -0.87451 -0.359088 -0.562066
-0.806806  0.455355 -0.376451 0.0120214
        0         0         0         1
Calibration result found
--- T_lidar_camera ---
 0.530073  0.143209 -0.835771  0.443286
-0.311332 -0.883928 -0.348918 -0.569202
 -0.78873  0.445154 -0.423961 0.0170462
        0         0         0         1
loading ouster_preprocessed/rosbag2_2023_03_28-16_25_54.bag.(png|ply)
loading ouster_preprocessed/rosbag2_2023_03_28-16_26_51.bag.(png|ply)

```
[image]

here's the calib file:
```
{
  "camera": {
    "camera_model": "plumb_bob",
    "distortion_coeffs": [
      -0.050701,
      0.111236,
      -0.000882,
      0.000141,
      -0.058891
    ],
    "intrinsics": [
      1454.658625,
      1455.212235,
      1229.262651,
      1010.66312
    ]
  },
  "meta": {
    "bag_names": [
      "rosbag2_2023_03_28-16_25_54.bag",
      "rosbag2_2023_03_28-16_26_51.bag"
    ],
    "camera_info_topic": "",
    "data_path": "ouster",
    "image_topic": "/image",
    "intensity_channel": "reflectivity",
    "points_topic": "/points"
  },
  "results": {
    "T_lidar_camera": [
      0.443286150136976,
      -0.56920243037414,
      0.017046161354092204,
      0.8423125580398597,
      -0.0498991391040071,
      -0.4821550932342098,
      0.23568219396750684
    ],
    "init_T_lidar_camera_auto": [
      0.3914099761630697,
      -0.5620658617641802,
      0.012021362595658946,
      0.8282034013794146,
      -0.04800531974072048,
      -0.5013323727522668,
      0.24584643021760905
    ]
  }
}
```
did i do something wrong?

koide3, 2024-07-03:
Possibly you forgot enabling dynamic points integration? See here: https://koide3.github.io/direct_visual_lidar_calibration/example/#2-preprocessing_1

khuechuong, 2024-07-03:
I added the -adv and it made it better. weird

closed 2024-07-03 by khuechuong