koide3/direct_visual_lidar_calibration #93: Result is off and wrong projection during calibration

khuechuong, 2024-07-03:
This is a follow up from issue [91](https://github.com/koide3/direct_visual_lidar_calibration/issues/91). I did with my dataset with a pretty wide angle lens. 
So I did intrinsic calibration with kalibr. The preprocessing cloud point looks good. The problem was that at the end, it was still kind of mismatching: 
[image]
I did see that during the calibration process, there was some mis-projection in the corner that projects my april tags to the wrong:
[image]

So I was wondering, how should I fix this? Should I go to an open environment with enough features like your rosbag?

koide3, 2024-07-04:
The viewer doesn't do hidden points culling, and thus this effect (back objects are colored by the pixels of front objects) itself doesn't mean a failure. Meanwhile, if you see misalignment on front objects, that means the calibration result is not satisfactory.

khuechuong, 2024-07-04:
So then, how would I fix it?