koide3/direct_visual_lidar_calibration #94: catkin_make error

kaifeng0306, 2024-07-08:
ubuntu18.04
CMake Error at /usr/local/lib/cmake/iridescence/iridescence-config.cmake:43 (add_library):
  add_library cannot create ALIAS target "Iridescence::Iridescence" because
  target "Iridescence::iridescence" is imported but not globally visible.

koide3, 2024-07-09:
Sorry, Ubuntu 18.04 has reached EOF, and we no longer support it. Please try to upgrade ubuntu.

duyingchang, 2024-07-23:
Is Ubuntu 20.04 OK？

koide3, 2024-07-23:
Ubuntu 20.04 should be ok. See the dockerfile for detailed instruction process.
https://github.com/koide3/direct_visual_lidar_calibration/blob/main/docker/noetic/Dockerfile

zhayuan11, 2025-01-03:
我在Ubuntu 20.04也遇到了同样的错误，请问怎么解决？

wzwan-developer, 2025-01-03:
> 我在Ubuntu 20.04也遇到了同样的错误，请问怎么解决？
[image]
[image]

katelord111, 2025-01-06:
> > 我在Ubuntu 20.04也遇到了同样的错误，请问怎么解决？
> > ![picture](https://private-user-images.githubusercontent.com/119391376/399933689-6c56715c-a37c-4ce4-9b52-353862d3ac24.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNTY3MjYsIm5iZiI6MTczNjE1NjQyNiwicGF0aCI6Ii8xMTkzOTEzNzYvMzk5OTMzNjg5LTZjNTY3MTVjLWEzN2MtNGNlNC05YjUyLTM1Mzg2MmQzYWMyNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQwOTQwMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MjIzYzE5NzQ2MjM4ZjhmNmM0ZWI5NmYzYWNkYzhhMjdmM2Q1NmIyZTA2ZDE5YWQwOTI4OGE2ZDQ2YWIyMTc1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.O1Jm4VPulhoV64-uLzDUkz9bTpEy-d28Ninn9O5DpOQ)
> > ![2025-01-03_191927_447](https://private-user-images.githubusercontent.com/119391376/399934240-3070427d-6d3a-42e0-a505-87c8e73b31ed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNTY3MjYsIm5iZiI6MTczNjE1NjQyNiwicGF0aCI6Ii8xMTkzOTEzNzYvMzk5OTM0MjQwLTMwNzA0MjdkLTZkM2EtNDJlMC1hNTA1LTg3YzhlNzNiMzFlZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQwOTQwMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yM2RiMjcxZTQ5MzMwMWMwMmU2ZjU5MWQ3YTVhMTU5NGNhNjVmNjkzY2Y4MzY5MzYxMGQwM2MwMjJjZWMzMTllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.XSOrzR1frbX2ZYzFbxjSEoYtQBV43DN_ev0gzXNUgFI)

set_property(TARGET Iridescence::iridescence PROPERTY IMPORTED_GLOBAL TRUE)