koide3/direct_visual_lidar_calibration #97: Wierd Lidar image

AbdullahJirjees, 2024-07-09:
Hello, 

Do you have any idea why my liadar scan is look like this?

I am using ROS2, GPU Nvidia A3000, CPU 11th Gen Intel® Core™ i7-11850H @ 2.50GHz × 16

[image]

koide3, 2024-07-10:
You might forget enabling dynamic points integrator https://koide3.github.io/direct_visual_lidar_calibration/example/#2-preprocessing_1.

AbdullahJirjees, 2024-07-10:
You were right I did not enable the dynamic point integrator, so I did enable it and here is the output results of the preprocessing step, any idea what I am missing?

[image]

BestVanRome, 2024-07-24:
@AbdullahJirjees have you had any progress on this one?

themightyoarfish, 2024-12-05:
I also have similar results on my own data, anyone got any clues?