koide3/direct_visual_lidar_calibration #102: Massive Lag when trying to visualize own recorded bags

BestVanRome, 2024-07-19:
**Before opening an issue**
If the issue is about build errors:
- [ ] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [ ] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [ ] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [ ] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [ ] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
After preprocessing, trying to start initial_guess and visualize, but there is a massive Lag. 

**To Reproduce**
See video and shared data.

**Expected behavior**
Fluent view of the visualisation, like in the examples you provided.

**Screenshots and sample data**

https://github.com/user-attachments/assets/ef82c63c-5b92-40ff-9478-bbd8d55cde18

https://github.com/user-attachments/assets/60f0717c-0cb3-42ce-b823-28056fc7ed4e

https://github.com/user-attachments/assets/984ed39f-87d8-49df-a17d-06dc64334468

https://github.com/user-attachments/assets/833d873b-fe26-41e0-957a-1bbfc8ebb890

https://github.com/user-attachments/assets/a0568627-842c-4e17-88d2-297ba51e1e2c

**Environment:**
 - OS: Ubuntu 22.04 (host)
 - [Your Ros Noetic Docker](https://hub.docker.com/repository/docker/koide3/direct_visual_lidar_calibration) 
 - ROS version: ROS1 noetic

**Additional context**
DATA to REPRODUCE:
2 of the bags and the preprocessed data created with your tool. -> [here](https://cloud.uni-koblenz.de/s/AgdQnBSy9cCPatw)

devvaibhav455, 2024-09-13:
Hi,
I faced the delay while using the docker container as well. But when I installed everything on the system without the docker, it works much smoother. However, I can't run superglue  as I don't have enough GPU memory and can't understand the manual point selection part. I can understand the pointcloud in rviz but it doesn't make sense to me in the visualizer available in the repo.

koide3, 2024-09-22:
The error stems from negative per-point timestamps and should be fixed by https://github.com/koide3/direct_visual_lidar_calibration/pull/116

BestVanRome, 2024-09-25:
I can't see a lot of difference. 
I exchanged your the old file with the new provided `time_keeper.cpp` inside my docker.   
In the video I click and drag just a second after the mouse is hovered over the screen.

https://github.com/user-attachments/assets/31e2611e-4175-4246-a7a4-8600e5118cda