koide3/direct_visual_lidar_calibration #107: STEREO CAMERA

ManChrys, 2024-08-25:
Hello @koide3 @Seekerzero  ,

thanks the amazing package ! 

i would like to ask if i use a stereo camera like zed2 whats the best topic should i use ?

Thanks in advance!!

The ZED node publishes data on the following topics:

    Left camera
        rgb/image_rect_color: Color rectified image (left RGB image by default)
        rgb/image_rect_gray: Grayscale rectified image (left RGB image by default)
        rgb_raw/image_raw_color: Color unrectified image (left RGB image by default)
        rgb_raw/image_raw_gray: Grayscale unrectified image (left RGB image by default)
        rgb/camera_info: Color camera calibration data
        rgb_raw/camera_info: Color unrectified camera calibration data
        left/image_rect_color: Left camera color rectified image
        left/image_rect_gray: Left camera grayscale rectified image
        left_raw/image_raw_color: Left camera color unrectified image
        left_raw/image_raw_gray: Left camera grayscale unrectified image
        left/camera_info: Left camera calibration data
        left_raw/camera_info: Left unrectified camera calibration data

    Right camera
        right/image_rect_color: Color rectified right image
        right_raw/image_raw_color: Color unrectified right image
        right/image_rect_gray: Grayscale rectified right image
        right_raw/image_raw_gray: Grayscale unrectified right image
        right/camera_info: Right camera calibration data
        right_raw/camera_info: Right unrectified camera calibration data

    Stereo pair
        stereo/image_rect_color: stereo rectified pair images side-by-side
        stereo_raw/image_raw_color: stereo unrectified pair images side-by-side

        Note: to retrieve the camera parameters you can subscribe to the topics left/camera_info, right/camera_info, left_raw/camera_info and right_raw/camera_info

    Depth and point cloud
        depth/depth_registered: Depth map image registered on the left image (32-bit float in meters by default)
        depth/camera_info: Depth camera calibration data
        point_cloud/cloud_registered: Registered color point cloud
        confidence/confidence_map: Confidence image (floating point values to be used in your own algorithms)
        disparity/disparity_image: Disparity image

Seekerzero, 2024-08-28:
You shall only calibrate the camera with the lidar using the raw image since the camera intrinsics were calculated based on the raw images, not the rectified ones. Hope this helps.

ManChrys, 2024-08-28:
Ok and I have to choose the right or left image right?

Seekerzero, 2024-09-03:
yep, once you calibrate one camera, you can then also calibrate the other using the ODE calibrated extrinsic values for the stereo camera.

ManChrys, 2024-09-03:
Thanks for the help my friend !! ✌️

closed 2024-09-03 by ManChrys

ashwinjoseph95, 2024-11-21:
Hi,
I was using my custom rosbag collected for carrying out camera lidar calibration. I am using a Zed2i stereo camera and velodyne VLP32 lidar.
Do I set the camera model as plumb_bob model or equirectangular?

Also now in the commands used for the calibration,I don't find any place to mention topic names for camera or lidar. Since i am using stereo camera i have 2 sets of camera image and info topics for both left and right in the rosbag, what should i do?