koide3/direct_visual_lidar_calibration #109: point cloud visualization

BC481, 2024-09-06:
**Before opening an issue**
If the issue is about build errors:
- [√] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [√] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/)

If the issue is about calibration procedure and results:
- [√] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [ ] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [ ] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
There is an issue with the point cloud visualization

**Expected behavior**
Hello, your work is great. I can successfully operate on Docker, but I encountered some problems when using the source code. When I am preprocessing the data, the visualization window crashes and shows a "Segmentation fault". If I do not perform visualization operations, the data processing can proceed normally.
When manually selecting points in the second step, a "double free or corruption (out)" error occurred.

**Screenshots and sample data**
[image]
[image]

**Environment:**
 - Ubuntu 20.04
 - ROS1 noetic

koide3, 2024-09-21:
The error may stem from the docker environment failing to access X11. Can you try running `xhost +` before launching the docker environment? Please also check if X11-related docker options are set as following: koide3.github.io/direct_visual_lidar_calibration/docker/

zhh2005757, 2024-11-05:
Please see my solution #123

koide3, 2024-11-10:
Thanks for sharing the solution @zhh2005757 .
I also think manually providing camera intrinsic parameters may help because it looks the program failed to find a camera_info topic.