koide3/direct_visual_lidar_calibration #111: Material of survey-grade high-reflectivity sphere targets when get reference the “pseudo” ground truth.

liunao123, 2024-09-12:
Hello, thank your excellent work.  

 When I test this calibration tool. I want to get the reference the “pseudo” ground truth too. In your paper, I found out that you just described it this way：As a reference, we estimated the LiDAR-camera transformation using survey-grade high-reflectivity sphere targets, as shown in Fig.7。

 So, I want to know **Material of survey-grade high-reflectivity sphere targets**, and **can you tell me how  can I make it**. I will try to get reference the “pseudo” ground truth by it. 
Thanks!

koide3, 2024-09-21:
We used this reflectance spheres: https://shop.laserscanning-europe.com/Laser-Scanner-Reference-Sphere-Set-Basic_1

liunao123, 2024-09-23:
@koide3  OK . I got it ， thank you very much.