koide3/direct_visual_lidar_calibration #113: Can't understand the dense pointcloud while doing the Initial guess (Manual)

devvaibhav455, 2024-09-13:
**Describe the bug**
It's a question actually. I can't understand the dense pointcloud and can't associate the points with the image. Someone please help in understanding it.

**To Reproduce**
Steps to reproduce the behavior:
1. I am using the ouster sample dataset that is provided in the documentation. 
2. I did the preprocessing step successfully and it generated me png and ply files from the ros2 bags
3. Try to do the initial guess (manual) (I can't do auto using superglue as I don't have enough GPU memory). It opens a photo and the dense point cloud. I can relate the points in the original ros2 bag visualized in rviz with the images in the preprocessed data but can't make sense of the dense point cloud. 
I tried moving the pointcloud around but can't understand anything from that as well.

**Expected behavior**
Shouldn't the dense point cloud look similar in shape to the original but just more points.

`ouster_results` contains the ply and png files.
`ros2 run direct_visual_lidar_calibration initial_guess_manual ~/Downloads/temp/ouster_results/`

**Screenshots and sample data**
[image]

In rviz, the cycle stand, the little light pole, trees are visible and can be associated easily. 

**Environment:**
Ubuntu 22.04 on amd64 architecture
ROS version: ROS2 humble

koide3, 2024-09-21:
Possibly, you forgot using the dynamic points integrator. Please see https://koide3.github.io/direct_visual_lidar_calibration/example/#ouster-camera-calibration

devvaibhav455, 2024-09-23:
Thanks. I think the `adv` option helped

closed 2024-10-14 by devvaibhav455