koide3/direct_visual_lidar_calibration #114: Colcon build error src/vlcal/calib/visual_camera_calibration.cpp:215:104: error: no matching function for call to ‘ceres::AutoDiffFirstOrderFunction<vlcal::MultiNIDCost, 7>::AutoDiffFirstOrderFunction(vlcal::MultiNIDCost*&)’

devvaibhav455, 2024-09-17:
**Before opening an issue**
If the issue is about build errors:
- [X] I have read [installation instructions](https://koide3.github.io/direct_visual_lidar_calibration/installation/)
- [X] I have considered using [docker](https://koide3.github.io/direct_visual_lidar_calibration/docker/) It gave me build errors while doing on arm64

If the issue is about calibration procedure and results:
- [X] I have read [data collection instructions](https://koide3.github.io/direct_visual_lidar_calibration/collection/)
- [X] I have confirmed [the program works on example data](https://koide3.github.io/direct_visual_lidar_calibration/example/)

If the issue is about the algorithm:
- [NA] I have read [the paper](https://staff.aist.go.jp/k.koide/assets/pdf/icra2023.pdf)

**Describe the bug**
Follow the instructions to install the pre-req except for absl (I had to build it as Position Independent Code as I faced this issue on a different PC).

```
~/calib_ws$ colcon build --cmake-args -DCMAKE_PREFIX_PATH=/home/nvidia/software/abseil-cpp/install
Starting >>> direct_visual_lidar_calibration
[Processing: direct_visual_lidar_calibration]                             
[Processing: direct_visual_lidar_calibration]                                     
[Processing: direct_visual_lidar_calibration]                                       
--- stderr: direct_visual_lidar_calibration                                         
** WARNING ** io features related to pcap will be disabled
/home/nvidia/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/calib/visual_camera_calibration.cpp: In member function ‘Eigen::Isometry3d vlcal::VisualCameraCalibration::estimate_pose_bfgs(const Isometry3d&)’:
/home/nvidia/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/calib/visual_camera_calibration.cpp:215:104: error: no matching function for call to ‘ceres::AutoDiffFirstOrderFunction<vlcal::MultiNIDCost, 7>::AutoDiffFirstOrderFunction(vlcal::MultiNIDCost*&)’
  215 | ew ceres::AutoDiffFirstOrderFunction<MultiNIDCost, Sophus::SE3d::num_parameters>(sum_nid);
      |                                                                                         ^

In file included from /usr/local/include/ceres/ceres.h:39,
                 from /home/nvidia/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/calib/visual_camera_calibration.cpp:7:
/usr/local/include/ceres/autodiff_first_order_function.h:125:12: note: candidate: ‘template<class ... Args, std::enable_if_t<is_constructible_v<vlcal::MultiNIDCost, Args&& ...>, void>* <anonymous> > ceres::AutoDiffFirstOrderFunction<FirstOrderFunctor, kNumParameters>::AutoDiffFirstOrderFunction(Args&& ...) [with Args = {Args ...}; std::enable_if_t<is_constructible_v<FirstOrderFunctor, Args&& ...> >* <anonymous> = <anonymous>; FirstOrderFunctor = vlcal::MultiNIDCost; int kNumParameters = 7]’
  125 |   explicit AutoDiffFirstOrderFunction(Args&&... args)
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/ceres/autodiff_first_order_function.h:125:12: note:   template argument deduction/substitution failed:
In file included from /usr/include/c++/11/bits/move.h:57,
                 from /usr/include/c++/11/bits/stl_pair.h:59,
                 from /usr/include/c++/11/bits/stl_algobase.h:64,
                 from /usr/include/c++/11/memory:63,
                 from /home/nvidia/calib_ws/src/direct_visual_lidar_calibration/include/camera/generic_camera_base.hpp:3,
                 from /home/nvidia/calib_ws/src/direct_visual_lidar_calibration/include/vlcal/calib/visual_camera_calibration.hpp:3,
                 from /home/nvidia/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/calib/visual_camera_calibration.cpp:1:
/usr/include/c++/11/type_traits: In substitution of ‘template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = void]’:
/usr/local/include/ceres/autodiff_first_order_function.h:124:69:   required from here
/usr/include/c++/11/type_traits:2579:11: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 2579 |     using enable_if_t = typename enable_if<_Cond, _Tp>::type;
      |           ^~~~~~~~~~~
In file included from /usr/local/include/ceres/ceres.h:39,
                 from /home/nvidia/calib_ws/src/direct_visual_lidar_calibration/src/vlcal/calib/visual_camera_calibration.cpp:7:
/usr/local/include/ceres/autodiff_first_order_function.h:116:12: note: candidate: ‘ceres::AutoDiffFirstOrderFunction<FirstOrderFunctor, kNumParameters>::AutoDiffFirstOrderFunction(std::unique_ptr<_Tp>) [with FirstOrderFunctor = vlcal::MultiNIDCost; int kNumParameters = 7]’
  116 |   explicit AutoDiffFirstOrderFunction(
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/ceres/autodiff_first_order_function.h:117:42: note:   no known conversion for argument 1 from ‘vlcal::MultiNIDCost*’ to ‘std::unique_ptr<vlcal::MultiNIDCost, std::default_delete<vlcal::MultiNIDCost> >’
  117 |       std::unique_ptr<FirstOrderFunctor> functor)
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/usr/local/include/ceres/autodiff_first_order_function.h:111:3: note: candidate: ‘ceres::AutoDiffFirstOrderFunction<FirstOrderFunctor, kNumParameters>::AutoDiffFirstOrderFunction(ceres::AutoDiffFirstOrderFunction<FirstOrderFunctor, kNumParameters>&&) [with FirstOrderFunctor = vlcal::MultiNIDCost; int kNumParameters = 7]’
  111 |   AutoDiffFirstOrderFunction(AutoDiffFirstOrderFunction&& other) noexcept =
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/ceres/autodiff_first_order_function.h:111:59: note:   no known conversion for argument 1 from ‘vlcal::MultiNIDCost*’ to ‘ceres::AutoDiffFirstOrderFunction<vlcal::MultiNIDCost, 7>&&’
  111 |   AutoDiffFirstOrderFunction(AutoDiffFirstOrderFunction&& other) noexcept =
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
gmake[2]: *** [CMakeFiles/direct_visual_lidar_calibration.dir/build.make:314: CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/calib/visual_camera_calibration.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:152: CMakeFiles/direct_visual_lidar_calibration.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< direct_visual_lidar_calibration [1min 43s, exited with code 2]

Summary: 0 packages finished [1min 43s]
  1 package failed: direct_visual_lidar_calibration
  1 package had stderr output: direct_visual_lidar_calibration

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the calibration workspace
2. Try to build the ROS2 workspace by specifying absl's path as its installed locally and not inside `/usr/local`.
`colcon build --cmake-args -DCMAKE_PREFIX_PATH=/home/nvidia/software/abseil-cpp/install`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots and sample data**
If applicable, add screenshots and sample data to reproduce your problem.

**Environment:**
 Ubuntu 22.04 on Jetson Jetpack 6.x (arm64)
 ROS version: ROS2 Humble

**Additional context**
The build process and pipeline worked on my laptop (amd64) once and I don't remember doing anything fancy except specifying the path and including the line `find_package(absl REQUIRED)` in the package's CMakeLists.txt

koide3, 2024-09-22:
Ceres introduced disruptive updates recently. For now, use a specific commit as indicated in [the instruction process](https://koide3.github.io/direct_visual_lidar_calibration/installation/). I'll update the code so that it can be built with the latest ceres.

devvaibhav455, 2024-09-23:
Build process now completes with a warning about opencv version issue

```
colcon build --cmake-args -DCMAKE_PREFIX_PATH=/home/nvidia/software/abseil-cpp/install -Dabsl_DIR=/home/nvidia/software/abseil-cpp/install/lib/cmake/absl
Starting >>> direct_visual_lidar_calibration
[Processing: direct_visual_lidar_calibration]                             
[Processing: direct_visual_lidar_calibration]                                     
[Processing: direct_visual_lidar_calibration]                                       
[Processing: direct_visual_lidar_calibration]                                       
[Processing: direct_visual_lidar_calibration]                                        
--- stderr: direct_visual_lidar_calibration                                          
** WARNING ** io features related to pcap will be disabled
/usr/bin/ld: warning: libopencv_core.so.4.5d, needed by /opt/ros/humble/lib/libcv_bridge.so, may conflict with libopencv_core.so.408
---
Finished <<< direct_visual_lidar_calibration [2min 32s]

Summary: 1 package finished [2min 33s]
  1 package had stderr output: direct_visual_lidar_calibration
```

And I get this error while doing preprocessing

```
ros2 run direct_visual_lidar_calibration preprocess /home/nvidia/Downloads/temp/ouster /home/nvidia/Downloads/temp/ouster_preprocessed -adv
data_path: /home/nvidia/Downloads/temp/ouster
dst_path : /home/nvidia/Downloads/temp/ouster_preprocessed
[INFO] [1727132316.535221821] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_26_51/rosbag2_2023_03_28-16_26_51_0.db3' for READ_ONLY.
[INFO] [1727132316.544416607] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
input_bags:
- /home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_26_51
- /home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54
[INFO] [1727132316.556413043] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
topics in /home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54:
- /camera_info : sensor_msgs/msg/CameraInfo
- /image : sensor_msgs/msg/Image
- /points : sensor_msgs/msg/PointCloud2
selected topics:
- camera_info: /camera_info
- image      : /image
- points     : /points
[INFO] [1727132316.563739538] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
intensity_channel: reflectivity
[INFO] [1727132316.580645415] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
try to get the camera model automatically
[INFO] [1727132316.625502036] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
camera_model: plumb_bob
image_size  : 2448 2048
intrinsics  : 1454.66 1455.21 1229.26 1010.66
dist_coeffs :   -0.0507013     0.111236 -0.000881838  0.000141199   -0.0588915
processing images and points (num_threads_per_bag=8)
start processing /home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54
[INFO] [1727132316.633343438] [rosbag2_storage]: Opened database '/home/nvidia/Downloads/temp/ouster/rosbag2_2023_03_28-16_25_54/rosbag2_2023_03_28-16_25_54_0.db3' for READ_ONLY.
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.8.0) /home/ubuntu/opencv_build/opencv/modules/core/src/matrix.cpp:246: error: (-215:Assertion failed) s >= 0 in function 'setSize'

[ros2run]: Aborted
```

devvaibhav455, 2024-09-23:
I solved the opencv issue using `export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libopencv_core.so.4.5.4d`