koide3/direct_visual_lidar_calibration #120: ubuntu20.04  arm  catkin_make error

keepermzw, 2024-10-11:
CMake Error at /usr/local/lib/cmake/iridescence/iridescence-config.cmake:43 (add_library):
  add_library cannot create ALIAS target "Iridescence::Iridescence" because
  target "Iridescence::iridescence" is imported but not globally visible.

How to solve this quetsion?please

santu97, 2024-10-12:
I also encountered this, have you solved it?

santu97, 2024-10-12:
I have solved it ：
in /usr/local/lib/cmake/iridescence/iridescence-targets.cmake
modify：add_library(Iridescence::iridescence SHARED IMPORTED) -> add_library(Iridescence::iridescence SHARED IMPORTED GLOBAL)

keepermzw, 2024-10-14:
> cmake

I have solved it with your advice,thanks

keepermzw, 2024-10-14:
> I have solved it ： in /usr/local/lib/cmake/iridescence/iridescence-targets.cmake modify：add_library(Iridescence::iridescence SHARED IMPORTED) -> add_library(Iridescence::iridescence SHARED IMPORTED GLOBAL)

I have solved it with your advice,thanks

closed 2024-10-16 by keepermzw