koide3/direct_visual_lidar_calibration #122: Question: Is it possible to mask out image regions?

themightyoarfish, 2024-10-28:
I would like to run this software in a scenario where some fixed or less fixed object is showing up in the camera and LIDAR data, which should be ignored for calibration purposes, e.g. something hanging down into the image.

Where would I add logic to filter out pixels or points so they do not participate in the optimization?

koide3, 2024-10-29:
I think such a mask can be implemented by modifying the [NIDCost](https://github.com/koide3/direct_visual_lidar_calibration/blob/e110ca230e9e262a93d7b2f90c2d498dbc970e71/include/vlcal/costs/nid_cost.hpp#L23)  class that measures the normalized information distance between a point cloud and an image.

BTW, I suppose just filling masked regions with a uniform color (e.g., black or white) should work in most cases because the NID effectively ignores such outliers (at least in theory).

themightyoarfish, 2025-07-29:
Although it's not a panacea, masking out more stuff, even if it is not moving during the recording, seems to also lead unsuccessful calibration.