koide3/direct_visual_lidar_calibration #123: Solution about error "double free or corruption (out)" in Ubuntu 18.04 environment

zhh2005757, 2024-11-05:
**Is your feature request related to a problem? Please describe.**
When running the repo codes in Ubuntu 18.04, it always reports "double free or corruption (out)" error. GDB finds out that the error is from "dataset.emplace_back(std::make_shared<VisualLiDARData>(data_path, bag_name));" in the codes, more specifically, from libiridescence.so.

**Describe the solution you tried**
This bug is due to the "-march=native" flag in Iridescence compilation. What we should do is turn on the cmake option "BUILD_WITH_MARCH_NATIVE" in CMakelists.txt of Iridescence repo and re-make (install) it. Then the viewer of this calibration program can work!

[image]

koide3, 2024-11-10:
Thanks for the detailed issue report. I guess you manually built and installed PCL (or some other libraries)? The error may stem from an inconsistent use of `-march=naive` option for dependency libraries. Mixing libraries built with/without `-march=native` causes segfaults. While "apt"-installed libraries are built without this flag, manual installation often enables it.

zhh2005757, 2024-11-10:
Yes I manually installed pcl 1.10 since the Ubuntu 18.04 built-in pcl library is 1.8. I think you are right that this error is caused due to the mismatching -march=native flags in different libraries.

koide3, 2024-11-10:
Thanks for confirming. I think this information is useful for people in the community.

XiaoHa0914, 2025-04-01:
> **您的功能请求是否与问题有关？请描述。** 在 Ubuntu 18.04 中运行 repo 代码时，它总是报告“double free or damage (out)”错误。GDB 发现错误来自代码中的“dataset.emplace_back(std::make_shared(data_path, bag_name));”，更具体地说，来自 libiridescence.so。
> 
> **描述您尝试过的解决方案** 此错误是由于 Iridescence 编译中的“-march=native”标志造成的。我们应该做的是在 Iridescence repo 的 CMakelists.txt 中打开 cmake 选项“BUILD_WITH_MARCH_NATIVE”并重新制作（安装）。然后此校准程序的查看器就可以工作了！
> 
> <img alt="旅游业_70" width="957" src="https://private-user-images.githubusercontent.com/38456100/383051763-3167ff99-8749-4b75-a6ce-be582376e346.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDM0Nzc3NzUsIm5iZiI6MTc0MzQ3NzQ3NSwicGF0aCI6Ii8zODQ1NjEwMC8zODMwNTE3NjMtMzE2N2ZmOTktODc0OS00Yjc1LWE2Y2UtYmU1ODIzNzZlMzQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDAxVDAzMTc1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlNzc0MWI4ZTg2YmQ2MWNhNDk2ODlmOGZhMzJjZWVjMTMyNDY0YWEzNTVmMjM1YTFhYzMxM2Y2YjczNzc1OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ZgbVoh7JDA12TY4nbZspAnpRTMip31xQBScUYhPFCoA">

请问我已经在iridescence/CMakeLists中打开了option(BUILD_WITH_MARCH_NATIVE "Build with -march=native" ON)，并且重新编译安装了，运行时依然是"double free or corruption (out)已放弃 (核心已转储)",我的环境是18.04,下载源码编译的PCL 1.10，请问应该如何做呢？
我的输入是：adc@adc:~/direct_visual_lidar_calibration$ rosrun direct_visual_lidar_calibration preprocess  /home/adc/rosbag /home/adc/direct_visual_lidar_calibration/src/direct_visual_lidar_calibration/result -adv
double free or corruption (out)
已放弃 (核心已转储）

zhh2005757, 2025-04-01:
我觉得你需要先用gdb确认一下报错是否来自于libiridescence.so或是其他地方 @XiaoHa0914

XiaoHa0914, 2025-04-01:
> 我觉得你需要先用gdb确认一下报错是否来自libiridescent.so或者其他地方[@XiaoHa0914](https://github.com/XiaoHa0914)
你好，我不确定我使用gdb是否正确，这是我的输入和输出：adc@adc:~$ ulimit -c 1000
adc@adc:~$ rosrun direct_visual_lidar_calibration preprocess  /home/adc/rosbag/ /home/adc/direct_visual_lidar_calibration/src/direct_visual_lidar_calibration/result -adv
double free or corruption (out)
已放弃 (核心已转储)
adc@adc:~$ gdb direct_visual_lidar_calibration preprocess core
Excess command line arguments ignored. (core)
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
direct_visual_lidar_calibration: 没有那个文件或目录.
/home/adc/preprocess: 没有那个文件或目录.
(gdb) 

麻烦您可以帮忙看一下吗