koide3/direct_visual_lidar_calibration #125: Throw segmentation fault (core dumped) while processing rosbag from sick lidar

Seekerzero, 2024-11-07:
Hi @koide3 , 

I was trying to calibrate a Realsense d435i with a sick multican 16 lidar using your calibration.
and it throws errors at the preprocess step.

I've checked the program was working on the other data (e.g. rosbag from livox lidar), so it should be a specific problem for the sick lidar. I tried to debug the problem and found out it happpend between the [loop ](https://github.com/koide3/direct_visual_lidar_calibration/blob/e110ca230e9e262a93d7b2f90c2d498dbc970e71/src/vlcal/preprocess/preprocess.cpp#L443), but not be able to nail it down further since it was randomly stopped between the lines. 

I've uploaded an example rosbag [here](https://storagex.seekerzero.science/s/z8uX).
And some additional details about the lidar data can be found [here](https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/sick_multiscan.launch). The intensity channel is named under the field "i".

Thank you so much

Seekerzero, 2024-11-08:
Hi Koide, 
I just found out it is caused by a typo in ros_cloud_converter. Please check the pull request.

koide3, 2024-11-10:
Ah, what a silly mistake I made! Thanks a lot for finding it! I just merged your PR!

Seekerzero, 2024-11-10:
No problem! My pleasure!

closed 2024-11-10 by Seekerzero