koide3/direct_visual_lidar_calibration #127: Shadow areas in point clouds without images

Ericshuai1, 2024-11-12:
Thank you very much for your outstanding project
[image]
In this image, there are shaded areas such as lane markings, making it easier to select points. However, in my image
[image]
There are no points in the selected point cloud that correspond to the image

koide3, 2024-11-12:
Maybe you can pick edges of windows or buildings?

supersupercoconut, 2024-11-17:
> Thank you very much for your outstanding project <img alt="image" width="300" src="https://private-user-images.githubusercontent.com/128591889/385208048-0b37185f-368d-445b-a010-c6fcf85d69ca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE4MTY0NzksIm5iZiI6MTczMTgxNjE3OSwicGF0aCI6Ii8xMjg1OTE4ODkvMzg1MjA4MDQ4LTBiMzcxODVmLTM2OGQtNDQ1Yi1hMDEwLWM2ZmNmODVkNjljYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExN1QwNDAyNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OWIyZTFlOGU0MDBjNzA1YmYyMDdkZWE5NGQ3ZGU0YmVkNDA2YjM3ZmZlMGZhZTdmMjY1ZGM3NWIxYTFlNzY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5febw6P6fmqls-pNgFg7aT-SIjExEmqqHg230d4O5P0"> In this image, there are shaded areas such as lane markings, making it easier to select points. However, in my image <img alt="image" width="941" src="https://private-user-images.githubusercontent.com/128591889/385208638-1cb8a98d-cd26-4c6e-9855-99d945240993.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE4MTY0NzksIm5iZiI6MTczMTgxNjE3OSwicGF0aCI6Ii8xMjg1OTE4ODkvMzg1MjA4NjM4LTFjYjhhOThkLWNkMjYtNGM2ZS05ODU1LTk5ZDk0NTI0MDk5My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExN1QwNDAyNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNjdkZjg2ZGI4ZjFmYmVkNDY5Zjg0ZmEzMTNkMzc5Y2Y0OGEzYzg2NDg0ZGE0Zjc3Y2I5YjU0MGY2OTUyYjQ3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.XEym9R8EtpC8gkfv5HUfdj9cmRV1OfUCMgr_JzKx-d0"> There are no points in the selected point cloud that correspond to the image

Why not select pointclouds by ourselves. I selected pointclouds again after preprocesss, and it works

koide3, 2024-11-17:
Pleaese note that you don't need to select points for every image. Picking 5 to 10 points through all imaged is sufficient.