koide3/direct_visual_lidar_calibration #128: Issue with pcl library while executing catkin_make

ashwinjoseph95, 2024-11-15:
Hi,
I was following the instructions to setup the tool using the instructions on https://koide3.github.io/direct_visual_lidar_calibration/installation/ 

Environment: Ubuntu 20.04 + ROS Noetic

After dealing successfully with the dependencies setup, I went ahead with the catkin_make command for direct_visual_lidar_calibration. I am getting some pcl library related undefined reference issues. Any help regarding how to fix it would be very helpful. Thank you.
My log looks like:

```
(venv3) ashwin@KN-Precision-7540:~/Ashwin/calib_ws$ catkin_make -j6
Base path: /home/knadmin/Ashwin/calib_ws
Source space: /home/knadmin/Ashwin/calib_ws/src
Build space: /home/knadmin/Ashwin/calib_ws/build
Devel space: /home/knadmin/Ashwin/calib_ws/devel
Install space: /home/knadmin/Ashwin/calib_ws/install
####
#### Running command: "cmake /home/knadmin/Ashwin/calib_ws/src -DCATKIN_DEVEL_PREFIX=/home/knadmin/Ashwin/calib_ws/devel -DCMAKE_INSTALL_PREFIX=/home/knadmin/Ashwin/calib_ws/install -G Unix Makefiles" in "/home/knadmin/Ashwin/calib_ws/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/knadmin/Ashwin/calib_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic;/usr/local;/opt/ros/noetic/share/random_numbers
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /home/knadmin/Ashwin/venv3/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home/knadmin/Ashwin/venv3/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /home/knadmin/Ashwin/venv3/lib/python3.8/site-packages/em.py  
-- Using empy: /home/knadmin/Ashwin/venv3/lib/python3.8/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/knadmin/Ashwin/calib_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
CMake Deprecation Warning at /usr/src/googletest/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googlemock/CMakeLists.txt:45 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/src/googletest/googletest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /home/knadmin/Ashwin/venv3/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
[... 316 lines cut ...]
  TARGETS
  direct_visual_lidar_calibration
)

install(
  TARGETS
  preprocess_map
  initial_guess_manual
  initial_guess_auto
  calibrate
  viewer
  DESTINATION
  lib/${PROJECT_NAME}
)

if($ENV{ROS_VERSION} EQUAL 1)
  # ROS1
else()
  # ROS2
  ament_python_install_package(${PROJECT_NAME})
  install(PROGRAMS scripts/find_matches_superglue.py DESTINATION lib/${PROJECT_NAME})
  ament_export_libraries(direct_visual_lidar_calibration)
  ament_auto_package()
endif()
```

koide3, 2024-11-24:
What is the version of PCL, and how you installed it (via apt?)?

harisbinyousaf24, 2024-12-26:
Its great to use this calibration tool @koide3. I successfully ran the tool using docker image and tested with my own data somehow. But i was trying to build this tool in my local system mentioned in the installation guide as mentioned by @ashwinjoseph95 and in the end step when i did catkin_make i got something like this:

`In file included from /usr/include/boost/detail/endian.hpp:9,
                 from /usr/local/include/pcl-1.9/pcl/PCLPointCloud2.h:11,
                 from /usr/local/include/pcl-1.9/pcl/pcl_base.h:55,
                 from /usr/local/include/pcl-1.9/pcl/surface/reconstruction.h:43,
                 from /usr/local/include/pcl-1.9/pcl/surface/convex_hull.h:47,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:12:
/usr/include/boost/predef/detail/endian_compat.h:11:161: note: #pragma message: The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead
   11 | #pragma message("The use of BOOST_*_ENDIAN and BOOST_BYTE_ORDER is deprecated. Please include <boost/predef/other/endian.h> and use BOOST_ENDIAN_*_BYTE instead")
      |                                                                                                                                                                 ^
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp: In function ‘double vlcal::estimate_lidar_fov(const ConstPtr&)’:
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:54:21: error: ‘make_shared’ is not a member of ‘pcl’
   54 |   auto cloud = pcl::make_shared<pcl::PointCloud<pcl::PointXYZ>>();
      |                     ^~~~~~~~~~~
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:54:21: note: suggested alternatives:
In file included from /usr/include/c++/9/memory:81,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/include/vlcal/common/frame.hpp:3,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/include/vlcal/common/estimate_fov.hpp:3,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:1:
/usr/include/c++/9/bits/shared_ptr.h:714:5: note:   ‘std::make_shared’
  714 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
In file included from /usr/include/boost/smart_ptr/make_shared.hpp:17,
                 from /usr/include/boost/make_shared.hpp:14,
                 from /usr/local/include/pcl-1.9/pcl/filters/boost.h:52,
                 from /usr/local/include/pcl-1.9/pcl/filters/voxel_grid.h:43,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:13:
/usr/include/boost/smart_ptr/make_shared_array.hpp:42:1: note:   ‘boost::make_shared’
   42 | make_shared(std::size_t size, const typename remove_extent<T>::type& value)
      | ^~~~~~~~~~~
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:54:62: error: expected primary-expression before ‘>’ token
   54 |   auto cloud = pcl::make_shared<pcl::PointCloud<pcl::PointXYZ>>();
      |                                                              ^~
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:54:65: error: expected primary-expression before ‘)’ token
   54 |   auto cloud = pcl::make_shared<pcl::PointCloud<pcl::PointXYZ>>();
      |                                                                 ^
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:58:24: error: ‘make_shared’ is not a member of ‘pcl’
   58 |   auto filtered = pcl::make_shared<pcl::PointCloud<pcl::PointXYZ>>();
      |                        ^~~~~~~~~~~
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:58:24: note: suggested alternatives:
In file included from /usr/include/c++/9/memory:81,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/include/vlcal/common/frame.hpp:3,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/include/vlcal/common/estimate_fov.hpp:3,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:1:
/usr/include/c++/9/bits/shared_ptr.h:714:5: note:   ‘std::make_shared’
  714 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
In file included from /usr/include/boost/smart_ptr/make_shared.hpp:17,
                 from /usr/include/boost/make_shared.hpp:14,
                 from /usr/local/include/pcl-1.9/pcl/filters/boost.h:52,
                 from /usr/local/include/pcl-1.9/pcl/filters/voxel_grid.h:43,
                 from /home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:13:
/usr/include/boost/smart_ptr/make_shared_array.hpp:42:1: note:   ‘boost::make_shared’
   42 | make_shared(std::size_t size, const typename remove_extent<T>::type& value)
      | ^~~~~~~~~~~
/home/ty/DVLC_ROS1/src/direct_visual_lidar_calibration/src/vlcal/common/estimate_fov.cpp:58:65: error: expected primary-expression before ‘>’ token
   58 |   auto filtered = pcl::make_shared<pcl::PointCloud<pcl::PointXYZ>>();
      |                                                                 ^~
[... cut ...]
make[2]: *** [direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/build.make:163: direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/src/vlcal/common/estimate_fov.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1593: direct_visual_lidar_calibration/CMakeFiles/direct_visual_lidar_calibration.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j4 -l4" failed

`
Just wondering is it related to my older version of pcl which is 1.9. Which version of PCL is required by DVLC?

koide3, 2024-12-29:
Yes, you need a PCL 1.2 or greater.

harisbinyousaf24, 2024-12-30:
Thanks @koide3. I worked it out with pcl 1.9 and by making some changes in the scripts estimate_fov.cpp and preprocess_map.cpp. The package built successfully. The preprocessing worked but if i activate --visualize flag i get this new thing Operation aborted/ Segmentation fault core dumped. Do i need to have GPU for the visualization?
Also would have been more easier if open3d is used as a visualization tool or some light library for non-GPU users (in my opinion). Because i had to pass initial guess manually via calib.json file to avoid point selection process which gets real stucky because i have no GPU.
Other than that really appreciate your work. @koide3

harisbinyousaf24, 2025-01-08:
Yeah sure @koide3 I am attaching a video in which i recorded my screen to run the tool:

https://github.com/user-attachments/assets/acbf3ab9-7c05-47c7-8264-4652654a055a

In this video if you see when i try to run this command:
`rosrun direct_visual_lidar_calibration preprocess --image_topic /Camera_8_synced/image_raw --points_topic /innovusion_lidar_front --camera_model plumb_bob --camera_intrinsic 1170.335979672321,1168.528989041779,957.0745338763222,543.629582572989 --camera_distortion_coeffs -0.3487691099377431,0.1406639011462764,0.0004765741803053816,0.0007804555751864781,-0.03061449907047344 /home/ty/Downloads/DATA/DVLC/data /home/ty/Downloads/DATA/DVLC/preprocessed --visualize`

I get the Segmentation Fault Core dumped as if something is revoking to open the visualizer. Although Camera info topic is not available, i am providing intrinsics and distortion coefficients separately. It works fine if i remove the --visualize flag from the end.

My System: Ubuntu 20.04, ROS Noetic
Processor: Intel® Core™ i5-7200U CPU @ 2.50GHz × 4 ;
Graphics: Mesa Intel® HD Graphics 620 (KBL GT2)

Built this repo from source.